Ultimaker/Cura #3400: [3.2.1] Middle wall stutter on cylindrical prints with 'Optimize Wall Printing Order' enabled

EngiNathan, 2018-02-25:
**Application Version**
3.2.1

**Platform**
Windows 10

**Display Driver**
nVidia GTX 1070 driver version 390.77

**Steps to Reproduce**
Load Cylindrical model in Cura > enable 'Optimize Wall Printing Order' > Slice object

**Actual Results**
Printer stutters while printing middle wall resulting in poor print quality.

**Expected results**
Middle wall should print smoothly

**Additional Information**
 Stuttering can also be seen in Cura's layer playback feature. No Stuttering occurs when 'Optimize Wall Printing Order' is not selected

[CFDMP_Cura test cylinder 100.curaproject.zip](https://github.com/Ultimaker/Cura/files/1756574/CFDMP_Cura.test.cylinder.100.curaproject.zip)

smartavionics, 2018-02-26:
I believe that this has already been fixed (see https://github.com/Ultimaker/CuraEngine/pull/695) but that hasn't yet been incorporated into Cura. Can you please confirm that you are using 3 or more walls which completely fill the cylinder wall (i.e. there's no infill in the cylinder wall).

tablatronix, 2019-12-02:
Is this the same problem I am having? I cannot get this to print concentric, pretty sure I used to in the past.. tried different settings, and wall sizes

[image]
[image]
[image]

smartavionics, 2019-12-02:
That ragged gap filling is due to the way that UM do the gap filling. If you install one of my builds the gap filling and thin walls is done differently and you will get a smooth result. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file you can find there.

tablatronix, 2019-12-02:
Thanks I will check it out, did this change at some point? I even tried concentric infill and it does not seem to make a difference.

smartavionics, 2019-12-02:
They have worked on the gap filling in the last couple of years but I don't think it has changed recently.