Ultimaker/Cura #3401: Individual mesh settings + smooth layer heights = sad cura

JJJpnt, 2018-02-26:
**Application Version**
3.2.1

**Platform**
Windows

**Display Driver**
NVidia's latest on a Gtx970

**Steps to Reproduce**
Slice with two meshes, one main and another inside of with mesh settings it to define a stronger infill zone. Add smooth layer height, press slice. 

**Actual Results**
Curaengine's stomach seems not to handle that mix and does an indigestion. Apparently tries to makes walls from the origin seeing the layer view, got a normal 1st layer, the second one is showing red wall lines from the origin to various places, noteworthy to say that the mesh that does change the infill isn't even contained in that layer. 

**Expected results**
Get a good slice of it ;)

**Additional Information**
Ok this is new and experimental, I just hope this information can help you :)
[image]

ianpaschal, 2018-02-26:
Thanks for reporting the bug!

Adaptive layers (that's what you mean right?) has a lot of limitations still. One of these is that it doesn't handle multiple objects very well, especially not using per-object settings.

This is something we're currently discussing, but it may involve saying for the time being that adaptive layers should just be used one object at a time.

We'll keep you posted.

ChrisRiddell, 2018-02-26:
I understand the Adaptive layers not supporting multi models because each model would not really want the same layer heights. 

But I hope Ultimaker will keep the mesh type models in mind as its good for doing custom infill and etc.

ianpaschal, 2018-02-26:
Ideally yes, this would be possible, but we've recently put another issue related to adaptive layers on the back-burner for similar reasons: it would require re-writing a _big_ part of Cura Engine. Since it's kind of an edge case using an experimental feature I don't imagine we'll get around to it any time soon.

I will discuss it with the Cura product owner and let you know if this is something we will put into the planning or leave for the time being.

BagelOrb, 2018-03-06:
For finding this issue later: "adaptive layers" "variable layer height"

Ghostkeeper, 2018-03-16:
I can't reproduce this issue. It seems to work fine.
[image]

As far as I'm concerned, adaptive layers should just work with multiple models since it's required for dual extrusion and because it should also work with a single model that has multiple parts.

ianpaschal, 2018-05-28:
Needs info for 2+ months and but went cold.

closed 2018-05-28 by ianpaschal