Ultimaker/Cura #3403: After Update old Material-Data are gon and backup can not be importetd.

ChaosLord76, 2018-02-26:
**Application Version**

3.2.1
**Platform**

Mac-OS
**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Simple update to this version
**Actual Results**

When i firt started the programm, i got an error saying unrecoverable issu. I was asked tohit the backup and reset button. I did so and after this the programm starts but without any configurations i did. also the ols material-Data can not be imported anymore (can not determin mime Typ)

**Expected results**

**Additional Information**

ianpaschal, 2018-02-27:
Hmm. I could have sworn I commented on this, but I guess I'm having deja vu.

The program has put your configurations into a zip file since one of them was what was causing the crash.

Can you share a log file from the crash?

Also, how are you trying to re-import your material data? Could you provide more specific steps to reproduce the problem?

ChaosLord76, 2018-02-27:
there was no backup file. not even a zip file, that is why i write her (sorry for my english, might be that i did not wrote it right)

For import i go to materials and import and select the file, i created with the last version bevor i updated the program.

I put the file, i got as export from the last version to this mail

hope that this might help you.

ianpaschal, 2018-02-27:
Sorry, you cannot attach files to a GitHub email. Paste the contents of the log between three backpacks like this:

\`\`\`
[Paste log here]
\`\`\`

weimarer76, 2018-03-10:
I have the same problem with 3.2.1 on Win10 x64.
Here is the backup file
[3.2_20180310_073749.zip](https://github.com/Ultimaker/Cura/files/1799183/3.2_20180310_073749.zip)
hope it helps

Ghostkeeper, 2018-03-16:
Argh, this one again. From the log:

    2018-03-10 07:37:31,666 - ERROR - [(8980)-MainThread] UM.Settings.ContainerStack.getConfigurationTypeFromSerialized [327]: Could not get configuration type: File contains no section headers.
    file: '<string>', line: 1
    '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\... <endless \x00's>

For some reason the files in your configuration got corrupt on disk. Cura is not able to recover them for you.

weimarer76, 2018-03-18:
Oh thanks!
I deleted the affected file "Ultimaker+2.global.cfg" and the update worked just fine.
regards!

closed 2018-03-18 by ChrisTerBeke