Ultimaker/Cura #3405: Filament Disapearing in the materials editor

mka84, 2018-02-26:
Filament Disapearing in the materials editor after i try to change the filament diameter from 2.85 to 1.75 wich renders the materials editor mute and makes to program hard to work with when changing settings. 
-->

**Application Version**
3.2.1

**Platform**
Win10

**Display Driver**
Gforce Game Ready Driver - 391.01 - 02/26/2018

**Steps to Reproduce**
Settings - Materials - Manage Materials - Duplicate - Change filament Diameter from 2.85 to 1.75 = Window close and the filament is gone 

**Actual Results**
Window closes and filament is gone

**Expected results**
Being able to change filament Diameter from 2.85 to 1.75 without the filament disapearing

ianpaschal, 2018-02-27:
@LipuFei @jackha This bug was found and fixed in your refactor, right?

Vandrasc, 2018-02-27:
This was fixed, we introduced a pop-up message informing the user that the diameter it is not supported, depending on the material, and the possibility to restore the value. It will be in the next release. See CURA-4947

closed 2018-02-27 by Vandrasc