Ultimaker/Cura #3406: Unable to slice above 0.35mm layer height.

Douggie-J, 2018-02-27:
**Application Version**
3.2

**Platform**
Windows 10

**Display Driver**
nvidia 388.13, GTX 1060 DX12

**Steps to Reproduce**
Change layer height to value above 0.35

**Actual Results**
Object is not sliced. Error message:
"Unable to slice with the current settings. The following settings have errors: Layer Height"

**Expected results**
Expected results are for the thing to work correctly.

**Additional Information**
Pic related. Have tried changing nozzle size, restarting after doing so. Issue does not appear to be present in 3.1. Doesn't seem to matter what model is loaded, even a simple cube will not slice.
[image]

Appesteijn, 2018-02-27:
What nozzle size do you use?

Douggie-J, 2018-02-27:
Setting any size of nozzle makes no difference, no matter how much larger than 0.35.

Amusingly, it will still happily slice 0.35mm layers with a 0.1mm nozzle.

Appesteijn, 2018-02-27:
If you use a Custom FDM printer, it will slice?

Douggie-J, 2018-02-27:
It does seem to work with a custom printer, actually.

BagelOrb, 2018-03-06:
@ianpaschal this is not an engine bug. It's about a setting becoming red, which makes the frontend not call the engine. It's probably somewhere in fdmprinter.def.json

BagelOrb, 2018-03-06:
@tylergibson decided to put a hard limit on the layer height.
The culprit is in the following line in malyan_m200.def.json :
`        "layer_height": { "minimum_value": "0.04375", "maximum_value": "machine_nozzle_size * 0.875", "maximum_value_warning": "machine_nozzle_size * 0.48125 + 0.0875",  "default_value": 0.13125 },`

Solving this 'problem' is not up to Ultimaker. We look forward to a pull request - if needed.

tylergibson, 2018-03-06:
The MP Select Mini cannot reliably print above .35mm layer heights, which is why there is no profile setting for it. The nozzle and heater cartridge cannot maintain temp above that layer height.

Douggie-J, 2018-03-07:
@tylergibson I don't know about the V1 or plastics requiring higher temperatures, but I had no trouble printing stuff with a 0.4mm layer height in PLA with a 0.8mm nozzle at 50mm/sec, 195C.

Also, @BagelOrb, am I just blind, or is there something there I'm missing? That snippet makes it look like the maximum value should be dependent on the nozzle size set in the machine `"maximum_value": "machine_nozzle_size * 0.875"`, but changing the nozzle size does not increase the maximum layer height.

Thanks for picking it out, either way. Changing the maximum value in the file manually provides a good workaround for me.

BagelOrb, 2018-03-08:
Yeah that does look weird. Are you sure you changed the setting Nozzle Size?

Douggie-J, 2018-03-08:
Yes, multiple times. Even closed Cura and opened it back up to make sure it stayed. Had no effect.

Changing the default nozzle size in the malyan_m200.def.json file here:

> "machine_nozzle_size": {
>            "default_value": 0.4,
>            "minimum_value": 0.15
>        },

Does increase the layer height maximum in proportion to the new value.

tylergibson, 2018-03-09:
@Douggie-J I see now.  The config files for the MP printers are all solely for using the default 0.4mm nozzle size.  Since Monoprice doesn't officially support other nozzle sizes, they aren't provided out of the box for the Cura profile.

To make changes yourself, you'll need to make the settings changes in the def.json, save it, clear the settings cache, then relaunch Cura and you should see updated values.
settings in AppData\Roaming\cura\3.2\definitions and
cache in AppData\Local\cura\3.2\cache\definitions\3.2.1
on Windows.

Ghostkeeper, 2018-03-20:
The inheritance from nozzle size doesn't work because the nozzle size is per-extruder and the layer height setting is global.

Our normal approach when a certain setting will clearly lead to bad results is to give a warning value (in organge) rather than actually preventing the user from slicing with that setting. I'll change the profile to make it a warning value instead of an error value.

Ghostkeeper, 2018-03-30:
I think we should consider that to be the fix then. It's more in line with Cura's normal behaviour to give warnings instead of preventing the user from doing something.

commit ae1ea9fc (2018-03-20): Change minimum and maximum layer heights into a warning

Because it's possible to try this, but the print result will probably be terrible. This keeps it consistent with the rest of Cura's behaviour.

Fixes #3406.
closed 2018-04-03 by Ghostkeeper