Ultimaker/Cura #3407: Enhancement: make reaction to mouse buttons editable

herrdeh, 2018-02-28:
I'm working with FreeCAD and blender and I'm used to a certain reactions to the mouse, in my case:
* mouse wheel => zoom object
* mouse wheel click & shuffle mouse => rotate object 
* shift & click mouse wheel => shift view

In FreeCAD, this is called "blender style", FC has a settings menu where I can adjust FC's reaction to different mouse button assignments. It would be great to have such a feature in Cura, as one may not spend as much time with it as with FC or blender, and it's quite confusing always to remember different reactions to the mouse at views which do look quite similar. (esp. for us over 50... (-;)

Aeroengineer1, 2018-05-05:
I too would very much like to see this improvement.  I use a gaming mouse where each time a new version comes out, I associate the mouse controls to a more typical cad system (the mouse control software has to know where the executable, and each new version installs in a new folder) .  It would be nice to have this natively.

Ghostkeeper, 2018-05-30:
Our project manager just removed this from our planning because we won't have time to implement it any time soon. Maybe later, but it has no priority for us.

Jorengarenar, 2023-07-05:
It still would be a great feature to have