Ultimaker/Cura #3408: Concentric 3D infill puts infill layers in mid air

bmackinnon2, 2018-02-28:
**Application Version**
3.2.1

**Platform**
Windows 10 - Concentric 3D infill puts layers into mid air. 

**Qt**
LGPLv3

**PyQt**
GPL

**Display Driver**

**Steps to Reproduce**
added concentric 3d infill pattern at 15% layer 84 and others throughout have layer of infill printed in mid air. 

**Actual Results**
Print fails

**Expected results**
smooth gradual concentric infill from top to bottom. 

**Additional Information**
[image]

ianpaschal, 2018-02-28:
Can you share your model so we can test what is going on at that point specifically? Or better yet, a whole project?

bmackinnon2, 2018-02-28:
I would gladly load the project, however I am not sure how to load it here?

ianpaschal, 2018-02-28:
If you've saved the .curaproject to your hard drive, you can upload it to a website like WeTransfer and paste the link here. We can then download it and take a look at why it's doing the weird stuff that it's doing.

CurtisLeeBolin, 2018-02-28:
@bmackinnon2, you can zip the file then drag and drop it into your comment.  Example below.

[example.zip](https://github.com/Ultimaker/Cura/files/1766100/example.zip)

bmackinnon2, 2018-02-28:
Perfect, thank you. Here is the project. 

[Concentric3d_light_head_github.curaproject.zip](https://github.com/Ultimaker/Cura/files/1767674/Concentric3d_light_head_github.curaproject.zip)

BagelOrb, 2018-03-06:
This problem is inherent to Concentric 3D infill.

Actually the normal concentric infill is limiting in a similar way.

They only work correctly for objects which don't vary that much layer per layer. Objects with mostly horizontal and vertical faces, like a standing cylinder, a mug or a building. Your models has rounded surfaces in the Z direction, which don't work well wit hconcentric infill types.

bmackinnon2, 2018-03-06:
The problem I mentioned happens continuously throughout the print at areas where indeed the model has vertical walls exactly like a mug or a cylinder.

BagelOrb, 2018-03-06:
The problem only occurs on layer 35, 62 and 84. The diagonal parts in the inside are between layer 8 and 65. It's only the stuff on layer 84 which I can't fully explain.

bmackinnon2, 2018-03-06:
Yes 84 to layer 85 is where it happens and also again at layer 110 and again something happens at layer 121 to 122. I would say that a is a pretty consistent problem throughout a mug like object.

BagelOrb, 2018-03-06:
Whatever problems there are are all **fundamental** to concentric infill problems.

If these problems are bothering you you should switch to a different infill type.

bmackinnon2, 2018-03-06:
Therefore, you just informed me there was a problem with the infill, but closed my issue? Why even have an infill that will fail on most of the prints in which it was designed for? I fail to see how this problem could not have a solution or that it is not worth finding a solution. The pattern puts ridges on the inside that are not joined on either side. It could have an undulating pattern that increases frequency with infill % or infill number of "walls." I dont understand how this pattern /\/\ printing from the bottom up is fundamental to this, could it not be /\ because that would solve what is happening.

BagelOrb, 2018-03-06:
I've explained the causes of the problems below layer 85. I don't think there are any problems above layer 85. Please make clear what the problems are at layer 110 and 120-122. I can then help you understand.

I don't understand your ASCII drawings or what you say about undulating frequency.

I don't know for how many objects the concentric infill options would fail, but people seem to like these options, so we keep them in. Custom settings are the users responsibility.

bmackinnon2, 2018-03-06:
The problem is the exact same problem that happens every single time the infill merges again on the top. I also whipped up a quick tube in fusion and put it into Cura with the same settings and got the same problems presenting themselves. 

[image]

[image]

[image]

BagelOrb, 2018-03-08:
That's not what's happening. Your first solution is already what's
happening.

Look at the layer view in line mode and disable viewing the shell.
Then go down through the layers and verify whether there is a line of
infill with no infill on the layer below it.

You'll see that the problem only exists below layer 90.

<https://user-images.githubusercontent.com/36910288/37063647-444c3b54-2157-11e8-95b2-e24c43f1d868.png>

[image: 20180306_154919]
<https://user-images.githubusercontent.com/36910288/37063687-74775f20-2157-11e8-9e3a-f9317e0ef9a3.jpg>

[image: 20180306_155226]
<https://user-images.githubusercontent.com/36910288/37063691-7ab47800-2157-11e8-888e-f20db51f9efa.jpg>

<https://github.com/Ultimaker/Cura/issues/3408#issuecomment-370960786>, or mute
the thread

BagelOrb, 2018-03-08:
Wait I don't get it. Why are we getting different results?

BagelOrb, 2018-03-08:
The results from your screenshot are not from the project file you sent.

The project file you sent had a layer height of 0.2, while the screenshot has 0.3mm.

bmackinnon2, 2018-03-08:
That is another problem I just realized, same as you, that was I was having. Whenever I save a Cura Project under File --> Save Project and then open said project it doesn't save the settings it just loads the settings of whatever I used when I previously had it open?

BagelOrb, 2018-03-08:
Perhaps the last time you opened a project file you opened it as model file
and told to remember your choice.

See under preferences what the default behaviour for project files is.

bmackinnon2, 2018-03-08:
It says "always open as a project"

BagelOrb, 2018-03-08:
That sounds like a very serious problem!

Could you maybe open a new bug report and clearly state an example of a
setting which is not saved?

It might depend on the setting or the profile or the machine configuration.

It's probably also good to attach the log file of when you saved the
project and when  you tried to load it.

Ghostkeeper, 2018-03-22:
All the issues that I've seen in this thread so far are indeed inherent to concentric infill types. The Concentric 3D infill type just varies the offset from the outer wall over time. It can't ensure that at all distances there will be a wall in all places, so for lots of models there will be pieces hanging in mid-air.

I'd advise to use Cross 3D in your case. It has many of the same properties as Concentric 3D but works better with more models.

BagelOrb, 2018-05-04:
I just checked the pattern again on a simple cube and I changed my mind.
I think we should indeed remove this infill setting.

Concentric was already not the best infill type with lots of situations in which it failed.
Concentric 3D has even more failing situations.
Even when the model has the same layers throughout it can fail.

I believe we shouldn't offer an infill structure which fails so often to our users.

[image]

BagelOrb, 2018-05-04:
I'm opening a new issue for this after all.

closed 2018-05-04 by BagelOrb