Ultimaker/Cura #3409: [3.2.1] Save Project - Error - There are no file types available to write with!

CurtisLeeBolin, 2018-02-28:
**Application Version**
3.2.1

**Platform**
Arch Linux, Not AppImage
https://www.archlinux.org/packages/community/x86_64/cura/

**Qt**
5.10.1

**PyQt**
5.10

**Steps to Reproduce**
Click `File` in the top menu
Click `Save Project...`

**Actual Results**
Error
There are no file types available to write with!

**Expected results**
A file save dialog to choose the name and location of the project file.

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/1766057/cura.log)
[cura.cfg.zip](https://github.com/Ultimaker/Cura/files/1766060/cura.cfg.zip)

fieldOfView, 2018-02-28:
Make sure libSavitar is installed: https://github.com/ultimaker/libsavitar

CurtisLeeBolin, 2018-02-28:
@fieldOfView, is libSavitar a new dependency with this release of Cura?

ianpaschal, 2018-02-28:
> is libSavitar a new dependency with this release of Cura?

It's not, but it is required to save Cura projects.

CurtisLeeBolin, 2018-02-28:
@fieldOfView and @ianpaschal, I built and installed libSavitar, it is now working correctly.  I'll file a bug report with Arch Linux so the maintainer can add a package for libSavitar and add the dependency to Cura.

Thank you for your help.

Building and install of libSavitar
```
$ tar -xf libSavitar-2.7.0.tar.gz

$ cd libSavitar-2.7.0/

$ mkdir build

$ cd build/

$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -G "Unix Makefiles" ..
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.4", minimum required is "3.4.0") 
-- Found PythonLibs: /usr/lib/libpython3.6m.so (found suitable version "3.6.4", minimum required is "3.4.0") 
-- Found SIP version: 4.19.7
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lee/Downloads/libSavitar-2.7.0/build

$ make
Scanning dependencies of target pugixml
[  5%] Building CXX object pugixml/CMakeFiles/pugixml.dir/src/pugixml.cpp.o
[ 10%] Linking CXX static library libpugixml.a
[ 10%] Built target pugixml
Scanning dependencies of target Savitar
[ 15%] Building CXX object CMakeFiles/Savitar.dir/src/ThreeMFParser.cpp.o
[ 21%] Building CXX object CMakeFiles/Savitar.dir/src/SceneNode.cpp.o
[ 26%] Building CXX object CMakeFiles/Savitar.dir/src/Scene.cpp.o
[ 31%] Building CXX object CMakeFiles/Savitar.dir/src/MeshData.cpp.o
[ 36%] Building CXX object CMakeFiles/Savitar.dir/src/Vertex.cpp.o
[ 42%] Building CXX object CMakeFiles/Savitar.dir/src/Face.cpp.o
[ 47%] Linking CXX shared library libSavitar.so
[ 47%] Built target Savitar
[ 52%] Generating python/sipSavitarpart0.cpp, python/sipSavitarpart1.cpp, python/sipSavitarpart2.cpp, python/sipSavitarpart3.cpp, python/sipSavitarpart4.cpp, python/sipSavitarpart5.cpp, python/sipSavitarpart6.cpp, python/sipSavitarpart7.cpp

Scanning dependencies of target python_module_Savitar
[ 57%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart0.cpp.o
[ 63%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart1.cpp.o
[ 68%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart2.cpp.o
[ 73%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart3.cpp.o
[ 78%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart4.cpp.o
[ 84%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart5.cpp.o
[ 89%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart6.cpp.o
[ 94%] Building CXX object CMakeFiles/python_module_Savitar.dir/python/sipSavitarpart7.cpp.o
[100%] Linking CXX shared library Savitar.so
[100%] Built target python_module_Savitar

$ sudo make install
[ 10%] Built target pugixml
[ 47%] Built target Savitar
[100%] Built target python_module_Savitar
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/lib/python3.6/site-packages/Savitar.so
-- Set runtime path of "/usr/lib/python3.6/site-packages/Savitar.so" to "/usr/lib64"
-- Installing: /usr/lib64/libSavitar.so.0.1.0
-- Up-to-date: /usr/lib64/libSavitar.so.0
-- Up-to-date: /usr/lib64/libSavitar.so
-- Set runtime path of "/usr/lib64/libSavitar.so.0.1.0" to "/usr/lib64"
-- Up-to-date: /usr/include/Savitar/ThreeMFParser.h
-- Up-to-date: /usr/include/Savitar/Types.h
-- Up-to-date: /usr/include/Savitar/SceneNode.h
-- Up-to-date: /usr/include/Savitar/Scene.h
-- Up-to-date: /usr/include/Savitar/MeshData.h
-- Up-to-date: /usr/include/Savitar/Vertex.h
-- Up-to-date: /usr/include/Savitar/Face.h
-- Up-to-date: /usr/include/Savitar/SavitarExport.h
-- Installing: /usr/lib64/cmake/Savitar/Savitar-targets.cmake
-- Installing: /usr/lib64/cmake/Savitar/Savitar-targets-release.cmake
-- Installing: /usr/lib64/cmake/Savitar/SavitarConfig.cmake
-- Installing: /usr/lib64/cmake/Savitar/SavitarConfigVersion.cmake
-- Installing: /usr/lib64/libpugixml.a
-- Up-to-date: /usr/include/pugixml.hpp
-- Up-to-date: /usr/include/pugiconfig.hpp
-- Installing: /usr/lib64/cmake/pugixml/pugixml-config.cmake
-- Installing: /usr/lib64/cmake/pugixml/pugixml-config-release.cmake
```

pumpal, 2018-08-09:
I have libsavitar  installed but am still getting this error.

kamushadenes, 2018-09-19:
I also have libsavitar installed on Arch and I am getting this error

Ghostkeeper, 2018-11-16:
We have no developers on Arch any more to test this, so there's no way to really reproduce it on this side. On Ubuntu, installing libSavitar using the default CMake settings lets it work fine.

CurtisLeeBolin, 2018-11-16:
I am just adding some information for completeness.

The Arch Linux package maintainer, Jelle van der Waa, has added libsavitar as a dependency, so it is installed automatically when cura is installed.

```
$ cura --version
/usr/bin/cura:48: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5
  linux_distro_name = platform.linux_distribution()[0].lower()
cura 3.4.1
```

Bellow are the warnings/debug/info/errors shown when opening an STL file then trying to save as a project.
```
2018-11-16 12:20:47,177 - WARNING - [MainThread] UM.Preferences.getValue [91]: Tried to get the value of non-existing setting info/send_slice_info.
2018-11-16 12:20:47,208 - DEBUG - [Thread-4] UM.Mesh.MeshData.calculateNormalsFromVertices [447]: Calculating normals took 0.0005714893341064453 seconds
2018-11-16 12:20:47,209 - DEBUG - [Thread-4] STLReader.STLReader.read [89]: Loaded a mesh with 3462 vertices
2018-11-16 12:20:47,212 - DEBUG - [Thread-4] UM.Mesh.MeshData.approximateConvexHull [403]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.00341033935546875 seconds. 122 input vertices. 122 output vertices.
2018-11-16 12:20:47,228 - DEBUG - [Thread-4] UM.FileHandler.ReadFileJob.run [64]: Loading file took 0.0 seconds
2018-11-16 12:20:47,232 - DEBUG - [Thread-4] UM.Mesh.MeshData.approximateConvexHull [403]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0032601356506347656 seconds. 122 input vertices. 122 output vertices.
2018-11-16 12:20:47,243 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [369]: Container [per_object_stack_117674689659736] added.
2018-11-16 12:20:47,261 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [369]: Container [per_object_stack_117674689828736] added.
2018-11-16 12:20:47,262 - DEBUG - [MainThread] cura.Machines.Models.BrandMaterialsModel._update [86]: Updating BrandMaterialsModel.
2018-11-16 12:20:47,276 - DEBUG - [MainThread] cura.Machines.Models.GenericMaterialsModel._update [23]: Updating GenericMaterialsModel.
2018-11-16 12:20:47,277 - DEBUG - [MainThread] cura.Machines.Models.BrandMaterialsModel._update [86]: Updating BrandMaterialsModel.
2018-11-16 12:20:47,290 - DEBUG - [MainThread] cura.Machines.Models.GenericMaterialsModel._update [23]: Updating GenericMaterialsModel.
2018-11-16 12:20:47,320 - WARNING - [MainThread] UM.Preferences.getValue [91]: Tried to get the value of non-existing setting info/send_slice_info.
2018-11-16 12:20:47,328 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/transparent_object.shader]...
2018-11-16 12:20:47,396 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-11-16 12:20:48,300 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.9s
2018-11-16 12:20:55,106 - WARNING - [MainThread] UM.Preferences.getValue [91]: Tried to get the value of non-existing setting info/send_slice_info.
2018-11-16 12:20:55,163 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [90]: There are no file types available to write with!
```

Also, the package has been flagged as out of date and hopefully the maintainer will be updating the package to version 3.6.0 soon.

In the mean time, I have been using version 3.6.0 Cura AppImage available on [Ultimaker's website](https://ultimaker.com/en/products/ultimaker-cura-software) with Arch Linux with no problems.  Projects save fine.

closed 2019-01-07 by nallath