Ultimaker/Cura #3415: Cura 3.2.1 does not set bed temp according to material profile

pbartko, 2018-03-01:
**Application Version**
3.2.1

**Steps to Reproduce**
Selecting Material from Material list to change printing temp and build plate temp.

**Actual Results**
Printing temperature is changed, but build plate temp is not changed.

**Expected results**
build plate temp should be changed when changing material

[image]

[image]

ianpaschal, 2018-03-02:
Pretty sure this would be fixed by the refactor, right @jackha @LipuFei ?

LipuFei, 2018-03-02:
@ianpaschal Fixed in the refactor

closed 2018-03-02 by ianpaschal