Ultimaker/Cura #3416: [3.2.1] Layer number over 999 are not completely visible

marekzehra, 2018-03-01:
I have problem with gcode, that is over 999 layers in Layer view. When sliding throw layers, last complete number visible is 999, 1000 is shown as 000. The last (first) digit is hidden.

**Application Version**
3.2.1

**Platform**
MacOS

**Steps to Reproduce**
Slice object to have 1000 layers and more.

**Actual Results**
Only 3 digits are shown in label.

Layer number under 999:
[image]

Layer number over 1000:
[image]

**Expected results**
All digits are shown in label 😉

Vandrasc, 2018-03-03:
I thought we fixed this, https://github.com/Ultimaker/Cura/pull/2540.

ianpaschal, 2018-03-03:
Sadly no.

In 3.2.1 `1` is invisible and `2` is cut off when they're in the thousands places. See below.

[image]

[image]

Ghostkeeper, 2018-04-16:
Doesn't reproduce for me. Perhaps it's MacOS only then, due to the fonts or the screen scaling.

ChrisTerBeke, 2018-07-30:
Seems fixed in https://github.com/Ultimaker/Cura/commit/b9a94a1025ee335027f0148a3a71e6e2512fbb75.  Please re-open if that's not the case.