Ultimaker/Cura #3417: Wiki Run Cura From Source Issues

ChrisRiddell, 2018-03-01:
OSX 10.13.3

I followed all the instruction to run Cura from source that's on the wiki here on the github down to a tee everything was going good until it came time to install libArcus,

I get these error's and can't work out why some basic googling lead me around in circles with no fix is their something missing from the wiki or is the wiki based for a different version of OSX.

> chrismbp:build chrisriddell$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2 -DCMAKE_PREFIX_PATH=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2 -DBUILD_STATIC=ON -DBUILD_PYTHON=ON -DBUILD_EXAMPLES=OFF ..
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3.4.0")
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPythonLibs.cmake:262 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/FindSIP.cmake:33 (find_package)
  CMakeLists.txt:23 (find_package)

-- Configuring incomplete, errors occurred!
chrismbp:build chrisriddell$

ianpaschal, 2018-03-01:
@LipuFei When you get a minute can you take a look at this?

pbannister, 2018-03-01:
Looks like I am going down roughly the same path - on OS/X 10.13.3 and wanting to run/build Cura from source.

FIrst, the instructions on Ultimaker/Cura, /cura-build, and /cura-build-environment seem to differ somewhat, especially around OS/X. Muddling through on a slightly different path.

Did just get libArcus built (or so it seems) using cura-build-environment. Whether this is a good path is unclear.

Any hints would be nice. :)

Shell psuedo-code of my current experiment:

```shell
# Install OS/X command line XCode tools.
xcode-select --install

# Install Homebrew if not yet.
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

# Possibly needed if you already have Homebrew installed.
brew update

# Install from Homebrew needed tools.
brew install python3
brew install pipenv
brew install cmake
brew install gcc

mkdir cura.build
cd cura.build
pipenv install --three
pipenv install numpy scipy colorlog netifaces zeroconf pyserial PyQt5==5.8.2
pipenv install sip protobuf # arcus

#
# Building Cura directly from the Ultimaker sources.
#
#git clone https://github.com/ultimaker/Cura.git
# ... skip to NO_COMMITS if using the virgin Ultimaker sources.

#
# Building Cura from a clone of the Ultimaker sources.
#
git clone https://github.com/pbannister/Cura.git
cd Cura && {
	# Add remote to upstream Ultimaker repository.
	git remote add upstream https://github.com/Ultimaker/Cura.git

	# Fetch all changes from upstream.
	git fetch upstream

	# Checkout master branch.
	git checkout master

	# Merge changes from upstream into master.
	git merge upstream/master

	# We are now up-to-date with Ultimaker.
}

# (Skip to here if NO_COMMITS.)

# #
# # Building Arcus directly from the Ultimaker sources.
# #
# git clone https://github.com/ultimaker/libArcus.git
# cd libArcus && {
# 	mkdir build && cd build {
# 		cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_STATIC=ON -DBUILD_EXAMPLES=OFF ..
# 		# ... TBD ... not working??
# 	}
# }

# #
# # Building CuraEngine directly from the Ultimaker sources.
# #
# git clone https://github.com/ultimaker/CuraEngine.git

#
# Building from cura-build-environment. (??)
#
git clone https://github.com/Ultimaker/cura-build-environment.git
cd cura-build-environment && {
	mkdir build && cd build && {
		source ../env_osx.sh
  		cmake .. -DCMAKE_BUILD_TYPE=Release
  		make

  		# ... TBD ... good so far?
	}
}

# ... TBD

```

ChrisRiddell, 2018-03-01:
I'm following this, https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-macOS

So I managed to get the cmake to work by adding these to the cmake command

-DPYTHON_INCLUDE_DIR=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include
-DPYTHON_LIBRARY=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/lib

but the make command fails with this 

> In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipArcuspart3.cpp:10:
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipAPIArcus.h:In file included from 10/Users/chrisriddell/Downloads/libArcus-master/build/python/sipArcuspart1.cpp:
:10:
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipAPIArcus.h/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include/python3.6m/sip.h::1032:
:10: /Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include/python3.6m/sip.h:fatal error32: :10: 
      'Python.h'fatal error : file not 
found      'Python.h'
 file not found
#include <Python.h>
         ^~~~~~~~~~
#include <Python.h>
         ^~~~~~~~~~
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipArcuspart0.cpp:10:
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipAPIArcus.h:10:
/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include/python3.6m/sip.h:32:10: fatal error: 
      'Python.h' file not found
#include <Python.h>
         ^~~~~~~~~~
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipArcuspart2.cpp:10:
In file included from /Users/chrisriddell/Downloads/libArcus-master/build/python/sipAPIArcus.h:10:
/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include/python3.6m/sip.h:32:10: fatal error: 
      'Python.h' file not found
#include <Python.h>
         ^~~~~~~~~~
1 error generated.
1 error generated.
make[2]: *** [CMakeFiles/python_module_Arcus.dir/python/sipArcuspart3.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/python_module_Arcus.dir/python/sipArcuspart1.cpp.o] Error 1
1 error generated.
1 error generated.
make[2]: *** [CMakeFiles/python_module_Arcus.dir/python/sipArcuspart0.cpp.o] Error 1
make[2]: *** [CMakeFiles/python_module_Arcus.dir/python/sipArcuspart2.cpp.o] Error 1
make[1]: *** [CMakeFiles/python_module_Arcus.dir/all] Error 2
make: *** [all] Error 2

ChrisRiddell, 2018-03-12:
I'll have to test it but might be worth adding to the wiki as this would effect all Mac's following this guide to the letter.

LipuFei, 2018-03-12:
@ChrisRiddell Could you check if you have any python header files in `/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include`? And also clean up your build directory and try again?

ChrisRiddell, 2018-03-12:
@LipuFei in `/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include` I only have 2 folders google and python3.6m 

in `/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/include/python3.6m` I only have a sip.h

also cleaning the build folder and tried again with command 

`cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2 -DCMAKE_PREFIX_PATH=/Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2 -DBUILD_STATIC=ON -DBUILD_PYTHON=ON -DBUILD_EXAMPLES=OFF ..`

give these error back again

> CMake Error at /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3.4.0")
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /Applications/CMake.app/Contents/share/cmake-3.10/Modules/FindPythonLibs.cmake:262 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/FindSIP.cmake:33 (find_package)
  CMakeLists.txt:23 (find_package)
-- Configuring incomplete, errors occurred!

ChrisRiddell, 2018-03-12:
@LipuFei I was able to build by making `-DCMAKE_PREFIX_PATH=/Users/chrisriddell/.pyenvs/3.6.4`

Also I don't think the wiki is correct for OSX 

Under [Arcus ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-macOS#arcus)I don't think you need to run 4. as its handled under the pyenv

under [Cura & Uranium ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-macOS#cura--uranium) 3 should be this should it not? as their is no /usr/local/bin/ folder on my install of OSX and when I compiled CuraEngine the terminal outputted `installing: /Users/chrisriddell/pyenvs/Python_3.6.4-PyQt_5.8.2/bin/CuraEngine` 

`$ ln -s [WORKSPACE]/[VIRTUALENV]/bin/CuraEngine .`

LipuFei, 2018-03-12:
@ChrisRiddell I recall that sometimes when I use `pyvenv` with `--copies`, the environment it creates doesn't actually have Python header files. You need to manually copied them there. Don't know if it is the case.

LipuFei, 2018-03-12:
I now use `pyenv` + `virtualenvwrapper`. That works pretty well for me.

LipuFei, 2018-03-12:
@ChrisRiddell Just checked the Wiki, I think a flag is missing in `mkvirtualenv`, I have updated that:
```
mkvirtualenv Python_3.6.4-PyQt_5.8.2 --always-copy
```

ChrisRiddell, 2018-03-14:
Closing as this is now resolved.

closed 2018-03-14 by ChrisRiddell

ianpaschal, 2018-03-14:
@ChrisRiddell You may have seen I was the one who originally wrote the wiki entry. To be honest, it's just a slightly formatted version of notes I wrote to myself while setting up my new computer when I started working at Ultimaker.

So, if you have the time, do you mind sharing any improvements to the instructions? I feel bad there was possibly misleading info in it but don't have the time to double check and test everything right now.

ChrisRiddell, 2018-03-14:
@ianpaschal if you want since I can edit the wiki, I can go through and make the changes needed? to make the guide fully work.

ianpaschal, 2018-03-14:
Yes that's what I mean. I don't really have time for the next few days at least to do it myself.