Ultimaker/Cura #3422: App crashes when going to layer view after preparing slice on large complex models

sam-david, 2018-03-02:
**Application Version**

2.6.66

**Platform**

Windows 10

**Qt**

**PyQt**

**Display Driver**

nvidia 1080 ti, 16gb of ram

**Steps to Reproduce**

This seems to happen when going to layer view after slicing any large complex model. Frequently large models will be fine to slice and prepare, but if you want to see layer view the app will crash with the backtrace above. I can give you a model to reproduce if necessary.

**Actual Results**

**Expected results**

**Additional Information**

###Error traceback
```
Version: 2.6.66
Platform: Windows-10-10.0.16299
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 191, in _render
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 138, in render
  File "C:\Program Files (x86)\cura-lulzbot 2.6\plugins\LayerView\LayerPass.py", line 95, in render
    batch.render(self._scene.getActiveCamera())
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 215, in render
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 246, in _renderItem
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGL.py", line 210, in createVertexBuffer
MemoryError

-------------------------
Thread 0x00002ea4 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x000042ec (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00003634 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00002480 (most recent call first):
[... 94 lines cut ...]
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x0000418c (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 946 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Current thread 0x00001510 (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 99 in show
  File "C:/buildbot/win-slave1/Cura2-windows/build/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 202 in event
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 202 in event
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 732 in run
  File "C:/buildbot/win-slave1/Cura2-windows/build/build/inst/bin/cura_app.py", line 70 in <module>
  File "C:\Python35\Lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "C:\Python35\Lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
```

ianpaschal, 2018-03-02:
Since version 2.6 is no longer supported, could you first please upgrade to the latest version and then see if the issue persists?

fieldOfView, 2018-03-02:
2.6.66 sounds like a recent build of the Lulzbot Edition. Their version of Cura has diverged quite a bit, to the point that we cannot support it here. Please go to https://code.alephobjects.com instead.

fieldOfView, 2018-03-02:
Having said that, it looks like this 32 bit version of Cura LE is running out of memory. This is one of the reasons Ultimaker only makes 64 bit builds.

Ghostkeeper, 2018-03-26:
It also seems like it's not your host memory running out but the GPU memory.

You can try going back to compatibility mode layer view (in the preferences). It uses less memory on the GPU, in theory about a third. But otherwise we won't be able to fix this for you.

closed 2018-03-26 by Ghostkeeper