Ultimaker/Cura #3423: [3.2.1] [BUG] Can't print another piece with USB

macaron16, 2018-03-02:
**Application Version**
3.2.1

**Platform**
macOS High Sierra

**Qt**
**PyQt**
**Display Driver**
NVIDIA GeForce GT 650M 512 Mo

**Steps to Reproduce**
after the first print on USB, I can't print another piece. 
I have a message "printer not available".

I have to click on "cancel the printing" + Yes and start the second print.
In the second try, it's works.

ChrisTerBeke, 2018-03-02:
So Cura doesn't recognize when the print is finished? Might have something to do with the firmware?

rflulling, 2018-03-05:
"No Printer Connected"
Seems to be the same bug. Cura 3.2.1 on 64 bit Windows 7 Pro, on i5-650, 8GB DDR3, 3TB SATA2, AMD R7-2XX 2GB DDR5. Controlling a "Power Spec i3 Mini (Wanhao)," over USB2. Setting is "Marlin." Though the custom skinned, Cura included used setting "RepRap (Marlin/Sprinter)." 

"No Printer Connected" after a print, looks like printer sleeps? 
"No Printer Connected" after sending an instruction printer doesn't like. Example: at max-Z (100) then press +Z a few more times. [note: only software knows 100 is max, hardware has no switch at max and will shudder if instructed to go beyond].
In some cases after a print, the Pause Abort buttons persist, regardless of button pressed, the "No Printer Connected" will pop up. 

Nothing done software side will restore this. Printer must be hard power reset. 

As ChrisTerBeke says, this could be a firmware issue, and might well be a part of the issue. I have considered. But there are no updates from Power Spec or Wanhao, though there are reports that other firmware can be used.

As Cura installs a full suite of Arduino drivers (that were never installed for my M3D Micro, or Micro Plus.) I must assume the core of the hardware is a cheep Arduino. 

Just discovered at time of this report that OEM skinned version of CURA has no issue in talking to printer even when newest version says "No Printer Connected," and despite this connectivity CURA 3.2.1 will not restore connection.

asantoni, 2018-03-13:
Happens to me on Windows too. Cura gets stuck after every print, with the Monitor tab showing "100%". If I click Abort, my printer rehomes (the normal Abort behaviour) and then I'm able to print again.

I don't think it's a Marlin issue, but I can't rule it out. I've been testing against their bugfix-1.1.x branch, one from a couple months ago, and one from yesterday. Same behaviour. I'm pretty sure this bug only started happening for me once I upgraded to Cura 3.0 though, or maybe 3.1. (3.2 has the bug too.)

asantoni, 2018-03-23:
I erased my end GCode, and replaced it with this stock end GCode from the Reprap wiki, and my problem went away:
`M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y100 F5000 ;move completed part out
M84 ;steppers off
G90 ;absolute positioning
`

I wish I had a backup of my old end g-code, but I don't. I don't think the bug is related to USB, though if this happens to someone else, see if you can reproduce it with printing off an SD card too. I think the bug is related to the end GCode somehow not triggering Cura to think the print is finished.

I had previously commented out a bunch of lines of my end GCode, trying to rule out the possibility it was related to that, but none of my changes fixed the problem. However, since completely replacing my GCode fixed the problem, then the problem is somehow related to the end GCode.

fieldOfView, 2018-03-23:
Do you have an old gcode file you can share?

ChrisTerBeke, 2018-05-23:
Now part of #3449.

closed 2018-05-23 by ChrisTerBeke