Ultimaker/Cura #3424: Suggestion: Balloon help for Profiles

Thisismydigitalself, 2018-03-02:
May i suggest adding a balloon help for profiles pull down menu as well (right pane print setup) for being able to see the complete name / description we gave it? I save profiles good for certain scenarios and try to name these with names that summarizes their settings. it will be nice if the balloon will break lines for easy reading based on '-' or '*' or whatever.

BagelOrb, 2018-03-06:
The profile names are limited in length. I can't make a profile name so long that it doesn't fit in the dropdown box: 
[image]

The full name shows when I open the dropdown box: "Super long profile name with a lot of ch" and the nI hit the max nr of chars.

Thisismydigitalself, 2018-03-06:
hmm... i did not know that and this obviously renders my idea useless.
How can I then quickly know which profile i should chose to use out of my dozen or so? 
i don't give names - i  give short description  like: PETG/230/75/LYR0168/IF20%/Overhangs/bla bla..

BagelOrb, 2018-03-06:
My settings are usually coupled to a specific project with its own models, so I just save the project file, which includes all settings.

Thisismydigitalself, 2018-03-06:
I do the same but I also have different profiles for say PLA and for PETG so i can switch back and forth to see which one suits my part better.

Ghostkeeper, 2018-03-26:
The length of the profile names are limited by the limit to the length of file names on your operating system, since we store these profiles by their names in the configuration folder.

We could add functionality to add a description to profiles for bookkeeping.

Thisismydigitalself, 2018-03-27:
Thank you. whenever i print a new challenging object i save its profile and name it so i can easily find and chose this profile again as a starting point for my next object with similar features and requirements. this saves me a lot of time and the outcome is just great.

Ghostkeeper, 2018-05-30:
Our project manager removed this from our planning. We won't get time to implement this any time soon, but maybe later.