Ultimaker/Cura #3426: Suggestions: Compatibility mode is really needed?

mtmello2003, 2018-03-02:
Hello,

My question is about the impossibility to use 3D resources from Cura, like gcode simulation for example.

 My windows 10 with intel video card (WDDM 1.1) always get "compatibily mode" in cura software but Cura really need better requirements of video card to allow user to use this feature???

 In my same computer, I can do the same (simulate gcode) using Slic3r, even with not best performance but It is up only to me! :)

My opinion is about Cura to reduce the requirement to open features like that and let users to use it or not, if the computer performance is not ok...

Thanks

ChrisTerBeke, 2018-03-02:
It's not so much about your GPU as it is about OpenGL support. Only OpenGL 4.1 and up (which is already quite old though) support the full simulation view. From OpenGL 2 onward we support the compatibility mode.

If you're having issues on a modern computer, try installing newer video drivers or update OpenGL.

BagelOrb, 2018-03-06:
I don't think Ultimaker shoukld invest time in getter a better experience for people with an outdated video card. The campatibility view is good enough.

Ultimaker will most probably never implement a third view.

closed 2018-03-06 by BagelOrb

erkserkserks, 2021-12-20:
For people coming here via a search engine:

It's possible to use a software OpenGL renderer as a workaround. On Windows, you can use llvmpipe from here: https://github.com/pal1000/mesa-dist-win

Unzip a release, e.g. mesa3d-21.3.2-release-mingw.7z, and run `perappdeploy.cmd`. It will ask where to install the software renderer to, point it to the Cura installation directory, e.g. `C:\Program Files\Ultimaker Cura 4.12.1`

My laptop's GPU (Intel HD Graphics 3000) only supports OpenGL 3.1, but I was able to use the software renderer to unlock all of Cura's rendering modes (layer thickness, speed, flow). Performance is pretty bad (~1 FPS), but it is at least possible to use these features.

Ghostkeeper, 2021-12-30:
There are 2 features of OpenGL4 we're using: Geometry shaders (for layer view) and triangle picking (for lay flat on face).

It's been suggested that we might get better performance by pre-computing the layer view geometry in the CPU. That way it doesn't get re-calculated at every frame. And as a bonus, we can keep that working with OpenGL2 as well. Then again, we should also just keep the VAO in VRAM (which was attempted for Cura 4.13 but initially failed).

The triangle picking is quite a bit harder to do. One trick could be to make a render with every triangle having a unique colour and then seeing which colour is under the cursor, but that's a bit difficult to do as well.

fieldOfView, 2021-12-30:
> One trick could be to make a render with every triangle having a unique colour and then seeing which colour is under the cursor, but that's a bit difficult to do as well.

And it would break with more than 16 million faces in the view, since the renderer can only return 8-bit per channel textures.

Ghostkeeper, 2022-01-11:
Yup, that sort of cases make it difficult. If the model has more than 16 million faces, you'd have to make 2 renders, each with a different set of colours. Nothing is impossible, but at some point it's good to just say that you need a newer version of OpenGL for it instead of rolling our own implementation.