Ultimaker/Cura #3428: Cura 3.2.1 Crashes on Mac OSX 10.12.6 when trying to browse plugins

rullywowr, 2018-03-03:
I was excited to try Octoprint integration, but cannot because I cannot browse plugins.

**Application Version**
3.2.1

**Platform**
Mac OSX 10.12.6

**Steps to Reproduce**
Go to Plugins -> Browse Plugins

**Actual Results**
Cura Crashes and exits to desktop

**Expected results**
I should probably see available plugins to download..

**Additional Information**
Attached Mac crash report

[Cura Crash Report - Mac OSX.txt](https://github.com/Ultimaker/Cura/files/1777009/Cura.Crash.Report.-.Mac.OSX.txt)

Vandrasc, 2018-03-03:
Did you also try with a clean install? And can you provide us the Cura log? Make sure to launch Cura, go to Help->Show Configuration Folder->cura.log. Thank you

fieldOfView, 2018-03-03:
Same crash mechanism as https://github.com/Ultimaker/Cura/issues/2617

ianpaschal, 2018-03-04:
Closing because it's a duplicate.

closed 2018-03-04 by ianpaschal