Ultimaker/Cura #3431: Ubuntu: object gets lost when zooming

herrdeh, 2018-03-03:
Ubuntu bionic, development branch, GNO'ME 3.27.90
Steps to reproduce:
1. zoom closer to your object (mouse or keys), keep focus besides the middle of your object
2. zoom until you no longer see your object, as it gets out of focus
3. zoom back (mouse or keys)
4. object should be visible again - but is not.

Vandrasc, 2018-03-03:
Thank you for reporting it! 
We had multiple tickets related to this issue but they were unfortunately set on low priority, it also occurs on os x and the only workaround is to restart Cura. We hope to address it as soon as possible. 
Devs FYI see CURA-2191

Ghostkeeper, 2018-03-28:
CURA-2191 is about adding a way to reset the camera, which has been implemented now.

This bug report is about zooming in so far that the camera sort of gets stuck. I can't reproduce it though...