Ultimaker/Cura #3434: [3.2.1] Maybe a bug?

mtmello2003, 2018-03-04:
Hello,

When I use this command at "start gcode" section:
M140 S{material_bed_temperature_layer_0}

The command M190 Sxxx from settings > Material > Plate temperature initial layer  is completely ignored...

ianpaschal, 2018-03-04:
Please share some examples/screen shots as well as providing other information such as what platform you're on, what printer you're using, etc.

mtmello2003, 2018-03-04:
Using win10 x64

When I set in Material > Build Plate temperature initial layer, gcode give a M190 Sxxx command:
[image]

[image]

But if I set M140 in start code, my setting in Material looks like is ignored:
[image]

[image]

fieldOfView, 2018-03-04:
That’s by design. Cura checks if your startgcode contains bed preheating commands. If it does, Cura won’t add preheat commands of its own. Same with extruder preheating commands

closed 2018-03-04 by ianpaschal

Ghostkeeper, 2018-03-28:
Maybe we should look for `M140` or `M190` commands instead?