Ultimaker/Cura #3435: Top Layer incomplete VERSION 3.2.1

mcdum, 2018-03-04:
**Application Version**
3.2.1

**Platform**
Linux - Xubuntu

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

[image]
[image]
[image]
[image]

https://www.dropbox.com/s/ek9ht6nplbzid78/TEST.curaproject.3mf?dl=0

fieldOfView, 2018-03-04:
Could you try the following: open the gcode in a decent texteditor, and put -say- 20 lines of dummy text to the end of the file, each line prepended with a ;

I have seen reports that some marlin firmwares eat the last couple of lines of the gcode file. Cura 3.2.1 does not always add the profile settings to the end of the gcode file.

Ghostkeeper, 2018-03-28:
The reports that you saw were specific to Ultimaker 2, fieldOfView. The bug was in the Ultimaker 2 display code for the main menu (yeah don't ask).

I am interested in knowing the difference in the g-code between these two prints. Maybe you could zip them up and post them here? Or provide a project file that can be opened in both Cura versions, created in Cura 3.1?

mcdum, 2018-03-30:
I send you a zip with g-code from Cura 3.1 and Cura 3.2 with project file. Difference in g-code in parts position because Cura 3.2 don't import correct position of parts from Cura 3.1.

Ghostkeeper, 2018-04-26:
Attachments via e-mail submissions to Github don't work. You'll have to upload an attachment via the website.

ChrisTerBeke, 2018-05-23:
No new info was added. Closing.

closed 2018-05-23 by ChrisTerBeke