Ultimaker/Cura #3437: No retraction on inner parts with 0% infill

Patralos, 2018-03-05:
**Application Version**
3.2.1

**Platform**
Windows

**Steps to Reproduce**
Print something like a vase without using spiralized mode (more wall thickness), therefore you need to set infill to 0%. 

**Actual Results**
Cura is not doing retraction for movement over parts, which should be infill.

**Expected results**
Retraction before moving over 0% infill parts.

**Additional Information**
[image]

ianpaschal, 2018-03-05:
Why do you expect retraction for moving over non-existing structures?

Patralos, 2018-03-05:
If there is no retraction, the material is stringing from side to side.

Ghostkeeper, 2018-03-05:
The assumption is that stringing on the inside of the model is not so bad because it's not visible. If your infill is visible (because you tried to emulate vase mode by setting the top thickness and infill to 0 or something) then this assumption is wrong.

You can get retraction back by setting Combing Mode to off. It won't try to find a path through the inside of the model then and just always assumes it has to retract then.

closed 2018-03-05 by Ghostkeeper

ctzsnooze, 2018-03-27:
Another example of why we need the retract while combing option