Ultimaker/Cura #3438: [3.2.1] Segfault at boot, crashes immediately

JPFrancoia, 2018-03-05:
**Application Version**
3.2.1-1

**Platform**
Archlinux

**Qt**
5.10.1-3

**PyQt**
5.10.1-1

**Steps to Reproduce**
Try to start Cura, however you want

**Actual Results**
Doesn't start

**Expected results**
Should start

**Additional Information**

I remove all the following files to start from scratch:

```
.config/cura
.local/share/cura
.cache/cura
```

Here is what I get in the terminal.

```
[(140263965377856)-MainThread] UM.Resources.__initializeStoragePaths [351]: Initializing storage paths
[(140263965377856)-MainThread] UM.Resources.__initializeStoragePaths [362]: Config storage path is /home/djipey/.config/cura/3.2
[(140263965377856)-MainThread] UM.Resources.__initializeStoragePaths [370]: Data storage path is /home/djipey/.local/share/cura/3.2
[(140263965377856)-MainThread] UM.Resources.__initializeStoragePaths [382]: Cache storage path is /home/djipey/.cache/cura/3.2
[(140263965377856)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[(140263965377856)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
[(140263965377856)-MainThread] UM.Qt.QtApplication.__init__ [88]: Detected most suitable OpenGL context version: 4.1 Core profile
[(140263965377856)-MainThread] UM.Qt.QtApplication.__init__ [104]: Command line arguments: {'debug': False, 'trigger_early_crash': False, 'external-backend': False, 'headless': False, 'disable-textures': False, 'qmljsdebugger': None, 'file': [], 'single_instance': False}
2018-03-05 14:42:07,673 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2018-03-05 14:42:07,691 - INFO - [(140263965377856)-MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [65]: Found CuraEngine at: /usr/bin/CuraEngine
2018-03-05 14:42:07,695 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2018-03-05 14:42:07,707 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2018-03-05 14:42:07,757 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2018-03-05 14:42:07,761 - INFO - [(140263202952960)-Thread-1] UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2018-03-05 14:42:07,769 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2018-03-05 14:42:07,771 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2018-03-05 14:42:07,774 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2018-03-05 14:42:07,778 - WARNING - [(140263965377856)-MainThread] STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2018-03-05 14:42:07,780 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2018-03-05 14:42:07,789 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2018-03-05 14:42:07,794 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2018-03-05 14:42:07,801 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2018-03-05 14:42:07,805 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2018-03-05 14:42:07,815 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2018-03-05 14:42:07,818 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2018-03-05 14:42:07,820 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2018-03-05 14:42:07,824 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalContainerProvider
2018-03-05 14:42:07,824 - WARNING - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2018-03-05 14:42:07,828 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2018-03-05 14:42:07,887 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2018-03-05 14:42:07,889 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2018-03-05 14:42:07,895 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2018-03-05 14:42:07,898 - INFO - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MonitorStage
2018-03-05 14:42:07,907 - WARNING - [(140263965377856)-MainThread] 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2018-03-05 14:42:07,913 - WARNING - [(140263965377856)-MainThread] 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2018-03-05 14:42:07,913 - ERROR - [(140263965377856)-MainThread] UM.PluginRegistry.loadPlugin [227]: Plugin 3MFWriter did not return any objects to register
[... 179 lines cut ...]
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f9191e30700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f9192631700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f91bfd4c540 (most recent call first):
  File "/usr/lib/python3.6/site-packages/UM/Qt/QtApplication.py", line 236 in initializeEngine
  File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 730 in run
  File "/usr/bin/cura", line 141 in <module>
[1]    18653 segmentation fault (core dumped)  cura
```

servingbaby, 2018-03-05:
I am able to confirm the crashing at start with this latest Cura update to 3.2.1-1.
Quick fix: Downgrading the "python-pyqt5 5.10.1-1" to "python-pyqt5-5.10-3" seems to correct the issue.
In my case, I used the cache: 
pacman -U /var/cache/pacman/pkg/python-pyqt5-5.10-3-x86_64.pkg.tar.xz

Simon-Swenson-8351, 2018-03-06:
I am having the same issue. I would post my console output, but it is basically the same as what was already posted.

compost-mantis, 2018-03-06:
Just confirming servingbaby's solution; downgrading python-pyqt5 5.10.1-1 to 5.10-3 corrected this issue for me.

xtendo-org, 2018-03-06:
I'm also having this issue. The console output is basically same as the OP. I built the current master version 9901780dbc33e2418c7a96ac3e8bdf8d786d0049 using the `cura-git` Arch package.

My platform:

```
$ uname --all
Linux system 4.15.6-1-ARCH #1 SMP PREEMPT Sun Feb 25 12:53:23 UTC 2018 x86_64 GNU/Linux
```

alekseisasin, 2018-03-06:
Seems like it is a bug related to QT version. 
DEVS: CURA-5055

slimjim2234, 2018-03-07:
Same issue here

archesys, 2018-03-09:
Same problem for me..

LipuFei, 2018-03-09:
@JPFrancoia Are you running from source code? I'm using ubuntu and have 2 environments, one with Qt 5.10 and the other with 5.10.1. 5.10.1 fails on me as well (segfault); 5.10 works. The Qts I'm using were compiled from source. I think there could be a bug in 5.10.1. I guess the solution for now is as @servingbaby described: downgrade to 5.10

JPFrancoia, 2018-03-09:
Nope, Cura was installed from the package manager. I updated my issue.

xdama, 2018-03-14:
Still happening to me and these people here https://www.reddit.com/r/archlinux/comments/84fzku/cura_segmentation_fault/

curiousbirds, 2018-03-22:
This also happens to me with the most recent `cura` and QT packages from the official Arch repositories.

thopiekar, 2018-03-22:
There is something fcuk'ed up in your packages either in Qt or something around it. The same crash happens when using Ubuntu Artful together with the PPA I maintain. I was looking for a time and couldn't find the reason, so I gave up and recommend users to upgrade to Ubuntu Bionic, which is going into beta in the following weeks. There cura works just fine.
So if it is really because of PyQt5, then take a look into Debian's patches. The PyQt5 package (5.9.2+dfsg-1build2) does the job here.

jelly, 2018-03-22:
@curiousbirds Usually commenting happens to me doesn't help, as you can see multiple people already reported
@thopiekar  The issue seems to be created by PyQt5, sadly PyQt5 doesn't have a git repository so it's not very trivial to bisect their changes and figure out what causes the segfault.

djpalumbo, 2018-03-26:
Out of curiosity, has anyone reported this to Riverbank Computing (creator of PyQt5)?

thopiekar, 2018-03-26:
@jelly: I still strongly believe (as long as nobody convinces me from something different) that the issues come from patches made by the Linux distributions. For example in Ubuntu there are the following patches applied:
```
thopiekar@home:~/Desktop/debian/patches$ LANG=C ls -la
total 32
drwxr-xr-x 2 thopiekar thopiekar 4096 Aug  22  2017 .
drwxr-xr-x 6 thopiekar thopiekar 4096 Aug  22  2017 ..
-rw-r--r-- 1 thopiekar thopiekar  936 Aug  22  2017 bytes_conversion.diff
-rw-r--r-- 1 thopiekar thopiekar 3872 Aug  22  2017 no_license_check.diff
-rw-r--r-- 1 thopiekar thopiekar 1041 Aug  22  2017 no_qml_debug.diff
-rw-r--r-- 1 thopiekar thopiekar 1355 Aug  22  2017 opengl_detection.diff
-rw-r--r-- 1 thopiekar thopiekar 3699 Aug  22  2017 qt59_support.diff
-rw-r--r-- 1 thopiekar thopiekar  102 Aug  22  2017 series
```
So what someone could do is disabling each patch one by one and see what causes the issue. The unpatched version Ultimaker uses in cura-build-environment just works fine.
But I agree it is a shame that Riverbank does not host any VCS server at all. Debian manages the code separately at https://salsa.debian.org/python-team/modules/pyqt5/commits/debian/master
However, I can also imagine the error comes from Qt5, too.
@djpalumbo: Nope, just because I'm not 100%ly convinced it is caused by their software.

LipuFei, 2018-03-27:
I'm using ubuntu, and this is the stack trace I get when I use PyQt 5.10.1 + Qt 5.10.1:

```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffecd55d8d in QWindow::geometry (this=0x1ccb6e0) at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtbase/src/gui/kernel/qwindow.cpp:1694
1694	        return QHighDpi::fromNativePixels(d->platformWindow->geometry(), this);
(gdb) py-bt
Unable to locate python frame
(gdb) bt
#0  0x00007fffecd55d8d in QWindow::geometry (this=0x1ccb6e0) at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtbase/src/gui/kernel/qwindow.cpp:1694
#1  0x00007fffecd5a5d5 in QWindow::width (this=<optimized out>) at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtbase/src/gui/kernel/qwindow.h:229
#2  QWindow::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at .moc/moc_qwindow.cpp:572
#3  0x00007fffee32b038 in QQmlPropertyData::readPropertyWithArgs (this=this@entry=0x7fff841eeb68, target=target@entry=0x1ccb6e0, args=args@entry=0x7fffffffbf90)
    at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/include/QtQml/5.10.1/QtQml/private/../../../../../src/qml/qml/qqmlpropertycache_p.h:308
#4  0x00007fffee32495d in QQmlPropertyData::readProperty (property=0x7fffffffbf70, target=0x1ccb6e0, this=0x7fff841eeb68)
    at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/include/QtQml/5.10.1/QtQml/private/../../../../../src/qml/qml/qqmlpropertycache_p.h:302
#5  loadProperty (v4=v4@entry=0x1c0ebf0, object=object@entry=0x1ccb6e0, property=...) at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/src/qml/jsruntime/qv4qobjectwrapper.cpp:139
#6  0x00007fffee325431 in QV4::QObjectWrapper::getProperty (engine=engine@entry=0x1c0ebf0, object=object@entry=0x1ccb6e0, property=0x7fff841eeb68, captureRequired=captureRequired@entry=true)
    at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/src/qml/jsruntime/qv4qobjectwrapper.cpp:261
#7  0x00007fffee325544 in QV4::QObjectWrapper::getProperty (engine=engine@entry=0x1c0ebf0, object=0x1ccb6e0, propertyIndex=6, captureRequired=<optimized out>)
    at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/src/qml/jsruntime/qv4qobjectwrapper.cpp:341
#8  0x00007fffee345fe5 in QV4::Runtime::method_getQmlQObjectProperty (engine=0x1c0ebf0, object=..., propertyIndex=<optimized out>, captureRequired=<optimized out>)
    at /home/lfei/workspace/projects/cura/packages/qt-everywhere-src-5.10.1/qtdeclarative/src/qml/jsruntime/qv4runtime.cpp:1496
#9  0x00007fff89e63312 in ?? ()
#10 0x0000000000000000 in ?? ()
(gdb) 
```

I checked the changes between PyQt 5.10 and PyQt 5.10.1 and I couldn't find anything that seems to cause this. I guess it's Qt 5.10.1.

f0lie, 2018-03-29:
For people who didn't cache the previous version of python-pyqt5 do this:
```
sudo pacman -U https://archive.archlinux.org/packages/p/python-pyqt5/python-pyqt5-5.10-3-x86_64.pkg.tar.xz
```
It does the same thing but downloads the previous version from the arch linux archives. I can confirm that Cura works by doing this.

jelly, 2018-04-01:
The issue is caused by a change in PyQt 5, reverting the following [change](http://pkgbuild.com/~jelle/0001-Revert-register-type-magic.patch) makes Cura start without a segfault. I've informed upstream by sending an e-mail to the mailing list.

archesys, 2018-04-02:
I fond a solution to bypass thé problèmes ans wait for another solution. I
run the .appimage ans it works.

jelly, 2018-04-04:
The latest development snapshot of PyQt seems to resolve the segfault,sadly since it doesn't use a VCS system I can't determine which change.

https://riverbankcomputing.com/software/pyqt/download5

MHaendel, 2018-04-06:
Same issue, Ubuntu 17.10

larsemil, 2018-04-10:
I had the same problem in Archlinux. But an upgrade of the system solved it. I guess the pyqt fix was in there.

jelly, 2018-04-23:
This ticket can be closed, it's a PyQt bug.

thopiekar, 2018-04-23:
True, but this bug tracker is still a good place to collect these bugs, because the community is widely located at different distribution and operation systems.

Additionally, it seems that the same patch needs to be applied at cura-vuild-environment. ( I think)

Ghostkeeper, 2018-05-11:
I'd still like to close this, because:
* There's nothing that needs to be fixed in Cura.
* Cura currently ships with PyQt 5.8.2 which doesn't suffer this problem. As a workaround you could use the official AppImage.
* We have an open issue laying around to upgrade to PyQt 5.10.3, which already includes the fix. We would not upgrade to 5.10.1 first now that 5.10.3 is out.

closed 2018-05-11 by Ghostkeeper