Ultimaker/Cura #3439: Cura EXTREMELY slow, takes 18% of CPU at idle

RChadwick7, 2018-03-05:
I just upgraded from Cura 3.x (Don't remember exactly) to 3.2.1. I've got a Monoprice Mini, running Windows 8.1 on a Core I7. As soon as Cura starts up, the CPU usage for Cura goes to about 18%, and stays there. I have a firewall blocking Cura. Even though I turned off all obvious options that would require Internet access, my firewall says Cura is trying to make a connection. Thinking this might have caused the issue, I disabled my firewall and restarted Cura, with the same symptoms. In addition to the 18% CPU usage, Cura is unusably slow. To simply open a file, I click on file (Which takes 4-7 seconds), then open file (Another 7-12 seconds), then I select the file (Another 2-4 seconds), then click load (12-20 seconds). I have been unable to rotate the model or do much else. The previous version was perfectly acceptable. I have tried uninstalling and reinstalling, with the same results.

alekseisasin, 2018-03-06:
@RChadwick7 Did you try before any beta Cura version.
 In fact the Cura 3.2.1 should work faster than previous version. Maybe you have some broken data from Cura in your user-data folder.

RChadwick7, 2018-03-06:
While I didn't try any beta versions, I think you were right about the data. I see that Cura now has a profile built in for the Monoprice mini, so I used that and removed the old profile, Now, CPU usage is normal, and the interface is very responsive. Suppose something in the old printer caused problems in 3.2.1

closed 2018-03-06 by RChadwick7