Ultimaker/Cura #3440: TweakatZ issue with Cura 3.2.1

reboz, 2018-03-05:
Cura 3.2.1, win 10
I tried to use TweakAtZ 5.1.1.
It worked fine last time I used it but now there's something wrong.
when I give only one Script it creates the right command in the G-code.
but when I try to insert 2 or more scripts everything changes.
instead of changing the Temperature (for example) from 220 to 200 at 5 mm and keep it for 25 layers it does this strage thing. 
At 5 mm it starts cooling down from 220 to 200 of 0.8° at layer ((220-200) / 25).
the result is that before the 5mm I have 220°, after the 5mm I have a gradual cooling untill the 5mm+25layers, instead of having suddenly 200 after 5mm.
LAYER:24
G0 X91.452 Y91.452 Z5
;TweakAtZ V5.1.1: executed at 5.00 mm
M117 Printing... tw@  5.0
M104 S219.800000 T0
LAYER:25
G0 X92.708 Y91.452 Z5.2
;TweakAtZ V5.1.1: executed at 5.20 mm
M117 Printing... tw@  5.2
M104 S219.600000 T0

alekseisasin, 2018-03-07:
Hi @reboz , Can you share you project, print screens of changes in scripts? It will help us to reproduce the scenario/story.

reboz, 2018-03-07:
last time I used this function I set like you can see in the image. but the result last time was that at 5mm of print the T° was reducted from 220 to 200 for 25 layers, while this time it reduces T° over the 25 layer of request in equal way (20° of drop in 25 layer so 0.8° a layer).
[image]
[image]
[image]

alekseisasin, 2018-03-12:
@reboz Actually you used wrong settings in the plugin. To reach expected result please do this:
[image]

closed 2018-03-15 by alekseisasin