Ultimaker/Cura #3441: [3.2.1] Dual extrusion - Prime Tower incomplete

J-charles-C, 2018-03-06:
**Application Version : 3.2.1
**Platform : OSX

**Steps to Reproduce**

Sorry  for my english.
I have an E3D Chimera head with on nozzle of 0.4mm and one of 0.8mm.
Parameters for the two extruders are the same.

When i select extruder one for large part and extruder two for detail, all is ok. 
[image]

But if i reverse, tower is not complete ans is mid air.
[image]

I have this problem with another STL dual color
https://www.thingiverse.com/thing:1681618

regards

BagelOrb, 2018-03-06:
Please attach your project file.

BagelOrb, 2018-03-06:
The prime tower should at least have a thickness of twice the nozzle size, so it should be 1.6.

closed 2018-03-06 by BagelOrb

J-charles-C, 2018-03-06:
I,

Exact. All is ok.  Many thanks.

regards