Ultimaker/Cura #3442: Extrusion Cool Down Speed Modifier crashes Cura 3.2.1

amagr0, 2018-03-06:
**Application Version** - CURA 3.2.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Select a dual extrusion printer, and modify the "Extrusion Cool Down Speed Modifier" to 2 ºC/s.

**Actual Results** - "CuraEngine.exe. has stopped working"

**Expected results** - Successfully slicing

[image]

BagelOrb, 2018-03-06:
Please attach your project file.

For an Ultimaker 3 the Extrusion cooldown speed modifier can't be above 1.6, because that is the value of the machine_nozzle_heat_up_speed, which is a machine setting which is not configurable by the user.

I cannot reproduce it.

amagr0, 2018-03-06:
[BEEVC.curaproject.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1784840/BEEVC.curaproject.curaproject.3mf.zip)

BagelOrb, 2018-03-06:
My Cura doesn't crash.

Note that an extrusion cool down speed modifier of 2 is highly unlikely, because the heatup speed is 2.
That means that the extruder is unable to heat up during printing ( 2 - 2 = 0 ).

Perhaps that setting should be limited to just under 2, rather than 2.

Note to other testers: you still need to manually set the Extrusion Cool Down Speed Modifier to 2.

alekseisasin, 2018-03-06:
I confirm it crashes in Cura 3.2.1. I checked your project in "development/latest" Cura version and it already works fine. I want to ask you to wait next Cura version.

alekseisasin, 2018-03-07:
@amagr0 I will close this issue, the fix already added.

diegopradogesto, 2019-09-10:
This was indeed fixed a while ago but this ticket was never closed.

closed 2019-09-10 by diegopradogesto