Ultimaker/Cura #3443: Bed temperature always goes to 130°C !!

teomadh, 2018-03-06:
Hello everyone, I ask you for help about a problem that is making me crazy.

First of all, I explain what my setup is:
I use a FlashForge Creator Pro printer with the Sailfish 7.8 firmware.
As slicer I use Cure 3.2.1 and to convert files from .gCode to .x3g I use GpxUi.
The materials I use are mainly PLA and PVA.

The problem:
My problem is that any print bed temperature I enter in the print file in Cura is not respected during Printing. The printer always bringing the temperature to 130 ° C, making the adhesion of the PLA insufficient.
I tried to set the printer by removing the check from "heated bed" but without success (as you can see from the attached images).
From somewhere on the internet I read that under the heading "gcode flavor" I had to put "Makerbot", but I ask you to confirm this.
Finally I also tried to override the print bed temperature using the GpxUi settings, but again without success. 
I must also specify, as you can see from the attached images, that in GpxUi I selected "ReplicatorG / MakerbotDesktop" as gflavor. Also here I have to ask you for a confirm. 

Thanks for your help.
Matteo
[image]
[image]

alekseisasin, 2018-03-07:
Hi @teomadh, Answering to your question about  "Makerbot" flavor, we cannot confirm that it will help you. Using flavors which are not intended can be dangerous for your printer, even we cannot test because it is Third Party printer. For this question contact manufacture of your printer. Maybe you should check technical documentation about your "bed temperature"

Ghostkeeper, 2018-03-30:
I'm wondering, if you save this file as .gcode, does it also generate a command to set the build plate to 130 degrees? That would be either `M190 S130` or `M140 S130` in the g-code file if you open it with a text editor. If the .gcode file works fine, then it's probably the GPX executable setting that build plate temperature.

I would indeed advise you to use Makerbot flavour for Sailfish firmware, but I can't confirm that this is also the case for the FlashForge Creator Pro.

JDRitchey, 2018-04-30:
I have a makerbot flavor 3d printer as well.  I have determined the source of the problem in the generated gcode file.  There appears to be a type-o in the first few commands for makerbot flavor gcode:

    ;FLAVOR:Makerbot
    ;TIME:2601
    ;Generated with Cura_SteamEngine 3.3.1
    T0
    M190 S90
    M104 S230
    M109 S230
    M82 ;absolute extrusion mode

should instead be
    T0
    M109 S90
    M104 S230
    M82

in makerbot flavor, the M109 command is for extruder, and will not do what is intended.
additionally, makerbot does not *necessarily* support M190 command.

source:
http://reprap.org/wiki/G-code

badguitar, 2018-05-24:
Just wanted to add that I have the same issue on my FFCP /GPX plugin and such. I solve the problem through the very much annoying method of resetting the temperature to what I want in octoprint right after the print starts. 

I have other misc. problems with dual extrusion where it will heat up nozzles to the wrong temperature or use the wrong one.

Ghostkeeper, 2018-07-05:
An easier workaround would be to add a post-processing script to change all `M109` commands to `M116` and another to change all `M190` commands to `M109`.