Ultimaker/Cura #3444: Missing layers in G Code

TyNiDesign, 2018-03-06:
Exporting from Solidworks 2014 as STL. When I preview layer view there are missing layers, the same thing happens when I print it.

I have tried changing the definition of the STL file and this has made no difference.
[image]

smartavionics, 2018-03-06:
Could you please make the STL available?

TyNiDesign, 2018-03-06:
Hi Mark,

 

Thanks for the quick reply.

 

Attached is the STL file.

 

I’m currently using Cura 3.2.1

 

Best regards,

 

Steve

 

email reply

 

This email has been checked for viruses by AVG.
http://www.avg.com

smartavionics, 2018-03-06:
Sorry, attachments don't work with emails to github, please send it directly to markb@smartavionics.com

smartavionics, 2018-03-06:
Hi Steve, I loaded that STL into FreeCAD and nothing was displayed and it reported a whole bunch of problems so I don't think Cura can make a good job of that.

[image]

TyNiDesign, 2018-03-06:
Thanks! I will take a look at freeCAD.

⁣Sent from Blue ​

alekseisasin, 2018-03-07:
@TyNiDesign , how is your model, did you find the issue?

TyNiDesign, 2018-03-07:
Yes fixed the problem. Solidworks was exporting a corrupt STL. I used FreeCAD to create the STL and all is now perfect.

⁣Sent from Blue ​

closed 2018-03-07 by alekseisasin