Ultimaker/Cura #3445: [3.2.1] Concentric 3D infill puts infill layers in mid air consistently throughout vertical cylinder with constant wall thickness

bmackinnon2, 2018-03-06:
Application Version
3.2.1

Platform
Windows 10 - Concentric 3D infill puts layers into mid air.

Qt
LGPLv3

PyQt
GPL

Display Driver

Steps to Reproduce
I was told my previous issue was specific to my model  and that concentric 3d infill was made for cylinders. So I whipped up a cylinder in fusion and put it into cura. Quite unexpectedly I got the exact same results as before. The infill is putting layers into mid air this can be seen throughout the model but for reference at layers:

185 and 186 

Using infill density of 15%, infill pattern of concentric 3d, infill overlap of 50%. 

Here is my attached Cura project file for this. Hopefully someone will be more helpful than my last issue on the same subject as this is clearly a problem for many models. 
[CCR10_mug_test.curaproject.zip](https://github.com/Ultimaker/Cura/files/1785662/CCR10_mug_test.curaproject.zip)

Actual Results
Print fails

Expected results
smooth gradual concentric infill from top to bottom without layers forming in mid air throughout the entire print. 

Additional Information
None.

alekseisasin, 2018-03-07:
HI @bmackinnon2, I see the edge case of this infill. It is difficult to find a very intelligent solution which works for all types of models. Did you try to use Cross 3D ?

bmackinnon2, 2018-03-07:
Yes actually that was my solution to the problem, wouldn't it be possible to have concentric 3d function in the same way, but with the concentric pattern instead of a cross? The cross 3d obviously doesn't have the error that the concentric 3d does.

alekseisasin, 2018-03-09:
For now I am not sure that we will make changes for this infill, because it's initial idea was work like now, maybe extend with different settings, this is probably, but coming time we do not have plan work on it.

Ghostkeeper, 2018-04-04:
This infill pattern doesn't work well on many models, but for many other models it will work all right.

I don't see a way in which we can make it work for straight vertical cylinders and still have it work in the general case, such as for a straight vertical cylinder with its hole off-centre.

ianpaschal, 2018-05-22:
Closing because it's a won't do. Discussion about concentric infill can be continued in #3774

closed 2018-05-22 by ianpaschal