Ultimaker/Cura #3446: [3.2.1] New Monoprice Select Mini printer profile has strange gaps.

ChrisHeerschap, 2018-03-06:
**Application Version**
3.2.1

**Platform**
MacOS High Sierra 10.13.3 - confirmed on recent macbook pro and older macbook air
 
**Qt**
Ultimaker download

**PyQt**
Ultimaker download

**Display Driver**
High Sierra default for Radeon Pro 455 2GB/Intel HD Graphics 530 1536MB

**Printer**
Multiple - using Prusa i3 profile with MPSM dimensions as well as newly added "Monoprice Select Mini v2" profile - bug is seen in the latter, not the former.

**Steps to Reproduce**
Open a simple part. Slice, select "Layer View", and investigate the edges and corners.

[mpsm-gaps.zip](https://github.com/Ultimaker/Cura/files/1786015/mpsm-gaps.zip)

- test-block-MPSMv2.3mf - using new MPSMv2 profile
- test-block-prusa-mpsm.3mf - using modified Prusa i3 profile
- test-block.stl - simple piece used in testing. (originally seen in other models as well)

**Actual Results**
the "Monoprice Select Mini v2" printer profile results in gaps at the corners, whereas the "Prusa i3" profile with dimensions modified to match the MPSM doesn't have these gaps.

**Expected results**
No gaps in the print.

**Additional Information**

Gaps in the print with the MPSMv2 profile:
[image]

No gaps using modified Prusa i3 profile:
[image]

Ghostkeeper, 2018-03-30:
It looks like this is the result of the Coasting setting. The Monoprice printers by default seem to have no coasting, but maybe you enabled coasting by accident?

ChrisHeerschap, 2018-03-30:
I'm seeing this when I first switch to the printer, and when I switch profiles within the printer. I didn't have any of the coasting options selected, but when I added them to my options, sure enough, the coasting was enabled. Removed the printer definition and re-added and Coasting is enabled again, so on my version of the printers at least - coasting is on by default. At least I know how to fix it, as unchecking Coasting turns that off.

ChrisHeerschap, 2018-03-30:
Coasting is definitely set in the profiles, if I switch profiles it asks me if I want to revert the settings, which includes turning coasting back on.

If coasting is actually desired, I could try printing parts with it enabled, I didn't do that as the preview looked strange.

Ghostkeeper, 2018-04-26:
Coasting may work better if you have a weak feeder, long Bowden tube or a material that is hard to retract. I suggest you try it!

From here it looks like this is expected behaviour then.

closed 2018-04-26 by Ghostkeeper

jontheepi, 2019-04-16:
I have gaps with the MP Mini V1 and V2 profiles.  Coasting is not enabled.  Any thoughts on how to fix?  I have re-installed Cura and recreated profiles, but nothing seems to work.

Ghostkeeper, 2019-04-17:
Could you share a project file that reproduces the problem?