Ultimaker/Cura #3448: [feature request] "Bottoms" for tree supports

nam37, 2018-03-06:
Application Version
CURA V 3.2
Platform
Windows

Qt
N/a

PyQt
N/a

Display Driver
NVIDIA

Steps to Reproduce
1 - Apply the "Tree supports" to any complex "small" (25mm square).
2 - Add a "Raft."

Actual Results
1 - When the tree supports are thin (Wall thickness = .4mm) bed adhesion is VERY poor unless a raft is added.

Expected results
The support should adhere to bed without needing the raft.

[image]

Details: I've printed many small and medium sized models using the new tree supports, and I've generally had very good results. However, unless I use a raft I often have the supports come off the bed. Normally I don't have ANY bed adhesion issues, only with tree supports and only the supports themselves.

I think a good solution here would be to add a couple/few bottom layers under the tree supports where they make contact with the print bed. This would give the benefit of the raft without adding the actual raft where it isn't needed.

Ghostkeeper, 2018-03-30:
We could fill the first layer where it touches the build plate. I see no harm in that.

I've usually been printing these with brim though. Then it attaches fine too.

Ghostkeeper, 2018-05-30:
Our project manager removed this from our planning because we won't have time to implement this any time soon. I guess it's not terribly important since a brim works just as well (but takes a bit more space).

mu2004, 2018-11-15:
A lot of people don't want to use brim/raft for optimal bottom smoothness, and the tress support without base adhesion is a major problem, rendering the tree support totally unusable. The same problem goes to the standard support that has small area touching the base too.
IMHO, any support should take care of the base by itself.

mbartlett21, 2019-05-29:
@mu2004 
You can use the "Enable support brim" setting in the support section which will create a brim under the support

TheNitek, 2019-10-15:
@mbartlett21 that only works when brim is enabled for the print, doesn't it?

Ghostkeeper, 2019-10-17:
No, the support brim can be enabled separately from the adhesion type. You could use a support brim together with a raft or skirt, and it'll basically do exactly what you need. So I'll consider this issue fixed.

closed 2019-10-17 by Ghostkeeper

njsokalski, 2024-05-09:
I think the problem usually occurs only when the Tree Support is relatively tall. What I would like is to be able to increase the base of the support without increasing the size of the entire trunk, therefore increasing the surface area for better adhesion.