Ultimaker/Cura #3450: Upgrade Issue 3.2 > 3.2.1

ArcAIN6, 2018-03-07:
**Application Version**

3.2.1-win64
**Platform**

Windows 7 64-bit

**Steps to Reproduce**

Have any version of Cura installed.
Download 3.2.1
launch 3.2.1 Installer
Installer asks to uninstall old version
Click yes
Installer uninstalls old version
Installer proceeds to install
Installer says that it cannot write cura.exe
Retry does not fix it.
Ignore does not leave a working install.
Abort Installer

**Actual Results**

Installer Log:
Output folder: C:\Program Files\Ultimaker Cura 3.2
Extract: Arcus.pyd... 100%
Can't write: C:\Program Files\Ultimaker Cura 3.2\Cura.exe

**Expected results**

Expected Cura to install. 

**Additional Information**

Forcing a completely download, uninstall, reinstall cycle is quite inefficient. In-place, incremental updates would be a much better solution. 

Solution: 
Exiting the installer after it has uninstalled older version, then relaunching the installer will result in a working install. I'm unsure as to what the internals of the installer is doing, but it appears to be attempting to re-use the directory structure of the uninstalled version of cura. Essentially, attempting to write files to directories that no longer exist due to the uninstall procedure.

Ghostkeeper, 2018-03-30:
It did successfully extract Arcus.pyd though. You're sure that Cura wasn't running while uninstalling/installing?

Ghostkeeper, 2018-03-30:
By default, old installations of Cura will have different folders where they install (Cura 3.1 gets installed in the 3.1 folder), so I wouldn't expect the uninstaller to have any influence over this.

fieldOfView, 2018-03-30:
Cura 3.2.1 gets installed in the Cura 3.2 folder though.

ArcAIN6, 2018-03-30:
@Ghostkeeper Cura was not running during the upgrade process. 

From what i can guess, it's looks as thought the installer doesn't wait for the uninstall process to complete, or the uninstall process is leaving stuff behind. 

I ended up uninstalling cura using the windows uninstaller, then checked to make sure everything was gone. after that, installing 3.2.1 went off without an issue.

Ghostkeeper, 2018-04-26:
The uninstaller will also fail to delete Cura.exe if it's still in use or otherwise locked. Maybe a virus scanner was quarantining it or something?

ianpaschal, 2018-04-26:
I run Windows 7 Pro 64 bit on my old 3ds max laptop at home. I'll test it over the weekend.

ChrisTerBeke, 2018-05-23:
No new info, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke