Ultimaker/Cura #3451: Dual Extrusion - Move extruder to safe position

amagr0, 2018-03-07:
**Application Version**
Cura 3.2.1

**Platform**
Win 10 Pro, 64 bit

**Display Driver**
NVIDIA GeForce 840M

**Printer**
Custom printer

**Steps to Reproduce**
Select a dual extrusion printer and generate the gcode.

**Actual Results**
After the last extrusion toolpath of one extruder, the extruder will stop on top of the 3D print, waiting for the other extruder to reach the _Initial Printing temperature_, this could be a 5 seconds wait, and will make a little hole on the print.

**Expected results**
Move the extruder to a secure place, away from the print.
If a prime tower is selected the "waiting position" could be inside the tower. For a "Ooze Shield" the position could be xx cm from outside of the shield.

[image]

[BEEVC_2.curaproject.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1790086/BEEVC_2.curaproject.curaproject.3mf.zip)

ChrisTerBeke, 2018-03-13:
For a custom FDM printer, these actions could be defined in the per-extruder start- and end-gcode. It would be nicer indeed if the default behaviour would be to wait inside the prime tower or another pre-defined coordinates (maybe even the home XY position).

amagr0, 2018-03-14:
Hello Chris, 
I'm actually using the end-gcode per-extruder to move to a save position (at the center of the "Prime Tower"), but when the temperature reaches the "Initial Printing Temperature" the extruder moves again to the top of the 3D print and then moves again to the "Prime Tower". 
The end-gcode solves the problem, but this situation could be more optimized.
[image]

BagelOrb, 2018-03-16:
This feature is already implemented. Enable extruder start/end pos abs and set the extruder start/end positions in the machine settings.

These are not user settings, but I gathered there was a PR by @fieldOfView in the pipeline which makes all machine settings use editable.

closed 2018-03-16 by BagelOrb

amagr0, 2018-03-21:
Hi @BagelOrb 
How can a I set the "Enable extruder start/end pos abs" ?

Tks

BagelOrb, 2018-03-21:
By changing the .def.json file of your printer. Take a look at how other
printer definitions work.

amagr0, 2018-03-21:
@BagelOrb 

Is this parameter that you are talking about?
"extruder_prime_pos_abs"

[image]

BagelOrb, 2018-03-21:
No that parameter is.for when an extruder first starts in a print. You need
machine_extruder_start_pos_abs or something.

All settings are listed in fdmprinter.def.json except some extruder
settings which are defined in fdmextruder.def.json

<https://user-images.githubusercontent.com/12901952/37708387-5fb9174c-2cfe-11e8-9301-77491bde21b2.jpg>

amagr0, 2018-03-21:
Ok, tks.

I found two parameters in the "fdmextruder.def" file:
"machine_extruder_start_pos_abs":
"machine_extruder_end_pos_abs":

AbeFM, 2018-05-15:
To make this work, you have to add the function definitions above (bool, etc), basically type in all the text pictured?
`#: fdmextruder.def.json
msgctxt "machine_extruder_start_pos_abs label"
msgid "Extruder Start Position Absolute"
msgstr ""`
Then this enables the options in the GUI and you set them there instead of hard coding into the JSON? It'll become: 
`#: fdmextruder.def.json
msgctxt "machine_extruder_start_pos_abs label"
msgid "Extruder Start Position Absolute"
msgstr ""
{ "label": "Absolute.....`
That sure doesn't jive with the rest of the file, or I don't get it.

I agree with the OP - this should automatically be set to use the center (not the "position" of the prime tower, which seems to be on the outer edge?). I also get the behavior where setting it as a tool change script has the toolhead going back to the previous location on the print after the change and before the prime tower/next extrusion.

-------------
It seems it could be here:
`#: fdmprinter.def.json
msgctxt "extruder_prime_pos_z label"
msgid "Extruder Prime Z Position"
msgstr ""

#: fdmprinter.def.json
msgctxt "extruder_prime_pos_z description"
msgid ""
"The Z coordinate of the position where the nozzle primes at the start of "
"printing."
msgstr ""

#: fdmprinter.def.json
msgctxt "extruder_prime_pos_abs label"
msgid "Absolute Extruder Prime Position"
msgstr ""

#: fdmprinter.def.json
msgctxt "extruder_prime_pos_abs description"
msgid ""
"Make the extruder prime position absolute rather than relative to the last-"
"known location of the head."
msgstr ""
`
from the file C:\Program Files\Ultimaker Cura 3.3\resources\i18n\fdmprinter.def.json.pot

In short I feel I'm not sure how to apply the information in this post to make the printer do nozzle changes at a user-defined position.

Ghostkeeper, 2018-06-13:
No, the file that you found is a translation template. Editing that will have no effect because the template is there only for translators to submit new translations of Cura.

To modify these settings you would have to find the extruders that you're using in `C:\Program Files\Ultimaker Cura 3.3\resources\extruders`. For a Custom FDM Printer that would be the files called `custom_extruder_1.def.json` up through however many extruders you have. Then in the `overrides` category above the `extruder_nr` override, add the following code:

            "machine_extruder_start_pos_abs": {"default_value": true},
            "machine_extruder_start_pos_x": {"default_value": 0},
            "machine_extruder_start_pos_y": {"default_value": 0},
            "machine_extruder_end_pos_x": {"default_value": 0},
            "machine_extruder_end_pos_y": {"default_value": 0},

So it will look something like:

        "overrides": {
            "machine_extruder_start_pos_abs": {"default_value": true},
            "machine_extruder_start_pos_x": {"default_value": 0},
            "machine_extruder_start_pos_y": {"default_value": 0},
            "machine_extruder_end_pos_x": {"default_value": 0},
            "machine_extruder_end_pos_y": {"default_value": 0},
            "extruder_nr": {
                "default_value": 0,
                "maximum_value": "7"
            }
        }

This would cause the printer to move to position [0, 0] before switching extruders. However it modifies your Cura installation permanently (until you reinstall Cura).

A neater solution would be to actually create a new printer definition but that is a bit more work.

amagr0, 2018-07-27:
Hello @Ghostkeeper,

I followed your sugestion with the overides codes, but the situation remains as I wrote on the first post.

[image]

Nothing is really happening during the extruder change, the extruder will stop on top off the print.
[image]

I normally add a code on the "Extruder End G-code" to move the extruder, but the extruder moves again to the top of the 3D print and then moves again to the "Prime Tower" position, as I wrote on my second post.

Is there any other way to solve this situation?

Thanks.

amagr0, 2018-08-20:
@smartavionics Could you please help me here? Even with the @Ghostkeepe suggestion I still have issues.  
Thanks

smartavionics, 2018-08-20:
Hi, I should say that I know almost nothing about dual extruder setups. Anyway, I put the following into the end gcode for both extruders and it moves the nozzle to the prime area before waiting for the temps.
````
M83
G1 F3000 E-3
M82
G92 E0
G0 F6000 X250 Y150
````

amagr0, 2018-08-20:
Hello @smartavionics thanks for your suggestion. I'm already use a G0 F3600 X160 Y175 on each  "Extruder End G-code", this only solve half the problem, since the extruder goes again to the top of the print (and leave some drip filament of the extruder that has increased the temperature) and then moves back again to print the "Prime Tower".

smartavionics, 2018-08-20:
Really? I didn't think it was going back to the print untll after it had done the warmup and prime. I attach the gcode I am getting.

[CFFFP_BEEVC_2.curaproject.gcode.zip](https://github.com/Ultimaker/Cura/files/2302662/CFFFP_BEEVC_2.curaproject.gcode.zip)

amagr0, 2018-08-20:
Is also happening in your g-code, after the M109, the extruder goes to the last printed position.
In the image bellow the extruder did not go back because it has to print the skirt, but during the remaing layers, it will always goes back to the print and then goes to the "Prime Tower" again.

[image]

smartavionics, 2018-08-20:
I have just realised that your setting of the extruder parameters above is wrong. You have:

https://user-images.githubusercontent.com/12901952/43321560-39ff3896-91a4-11e8-8c9a-9f04cc9f0fdd.jpg

But what you need is

````
    "overrides": {
        "machine_extruder_start_pos_abs": {"default_value": true},
        "machine_extruder_start_pos_x": {"default_value": 160},
        "machine_extruder_start_pos_y": {"default_value": 175},
        "machine_extruder_end_pos_x": {"default_value": 160},
        "machine_extruder_end_pos_y": {"default_value": 175},
        "extruder_nr": {
            "default_value": 0,
            "maximum_value": "7"
        }
    }
````
Your brackets are wrong!

smartavionics, 2018-08-20:
I don't know what is going wrong but I got it to print some diagnostics and the extruder start position absolute value is always false and those extruder start coordinates are all zero. So I don't think the above extruder overrides are working. Maybe those values are not saved in the project file?

amagr0, 2018-08-21:
Hello @smartavionics, how can I also confirm that the overrides are working?

smartavionics, 2018-08-21:
Sorry, I am not really familiar with how the configuration part of Cura functions. I think someone else like @BagelOrb or @Ghostkeeper could answer your question.

amagr0, 2018-08-21:
Thanks again for your time @smartavionics .

BagelOrb, 2018-08-21:
`extruder_nr` doesn't have a closing bracket.

BagelOrb, 2018-08-21:
Your spacing conventions don't follow the same as our .def.json files. If you put `{` always on a new line and at the same indentation level as the '}' you can see very easily if you did something wrong. That's why we do it that way.

amagr0, 2018-08-21:
Hello @BagelOrb thanks for your attention.
I did a Cura 3.4.1 clean install, modify custom extruders 1 and 2 files, and nothing really happens.
I generate a gcode without the modifications, and another with custom extruders mods, and both are the same.

In the image below you can see what is still happening.

[image]

I send in attachment the CuraProject, the custom extruders files, and the gcodes with and without the @Ghostkeeper suggestions.

Really appreciate your help.

amagr0, 2018-09-14:
Hello @fieldOfView, since the suggestions by the @Ghostkeeper did not solve what I describe in the issue, I tried to use your PrinterSettings Plugin and select the "Absolut Extruder Prime Position" checkbox, but unfortunately the results remains the same, as you can see in the image below.

[image]

Could you please confirm that you also have this issue?
Thanks

[Absolut_issue.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2383640/Absolut_issue.curaproject.3mf.zip)

amagr0, 2018-10-19:
Resolved on 3.5.1!
Now this issue is officially closed ;)

AbeFM, 2018-10-20:
Could you tell me how to do this now? I don't see anything in the regular settings for head position - are we to modify the printer definitions, or use start/end gcode...?

I started over with clean definitions (CR-10s and added extruder), right now the transition happens on a small first layer island which melts the part and takes it with the extruder when it moves.