Ultimaker/Cura #3452: Suggestion: Adaptive Retraction Distance

DerEchteLanger, 2018-03-07:
I notice with my printer that i need only tiny retracts for covering small distances, e.g. 0,5mm retract for 2cm travel. But for long travel moves (>5cm) i have to increase it to more than 1mm, which is unneccessary for shorter distances.

So it would be awesome to have a retract thats dependant on the distance of the travel move, maybe with 3 steps we can define ourself:
short retract: travelmove shorter than X cm -> A mm retract
medium retract: X < travel move < Y -> B mm retract
long travel: Y < travel move -> C mm retract

cmatthes, 2018-03-07:
Need to think about it. At first thought it sounds quite useful and would probably be quiet "easy" to implement!

ChrisTerBeke, 2018-03-08:
Actually it might be quite difficult, as layers in CuraEngine don't have any information about other layers. First thing CuraEngine does is define and slice the layers, then it does toolpath generation, skin expansion, etc on a per layer (2D) basis. Still doesn't mean we shouldn't try of course if it's a good addition.

DerEchteLanger, 2018-03-08:
Why would the feature require inter-layer information? It just needs to calculate the distance traveled for each travelmove and apply a retraction feature based on that.

Ghostkeeper, 2018-03-30:
Yeah I agree, implementing this won't be very difficult.

We should always be careful not to make the whole system too complicated though. If a user adjusts the retraction distance we don't want him wondering why it's not changing the retraction distance at certain spots, only to figure out later that it's because the travel moves are too short.

We've had some success here at Ultimaker with retracts that keep on retracting slowly during the travel move. This not only retracts further on long moves by nature, but also maintains a steady underpressure. I personally think that that idea has more promise in the end.

cmatthes, 2018-05-27:
Finally getting around to reply :) 

`We should always be careful not to make the whole system too complicated though.`

I fully agree! In this specific case I think it could be managed quite well by using the show settings behaviour already used by Cura for some settings: 
[image]

I thought of maybe simply calling it : "Split Retraction Settings " or "Split Retraction behaviour"
This setting could be disabled by default and therefore wouldn't confuse a user. 

After reading those two blog posts ( [Investigate Pressure in Extrusion system](https://somei3deas.wordpress.com/2018/01/15/an-attempt-to-investigate-pressure-in-the-extrusion-system-with-a-diamond-hot-end/) & [Some musings on retracts](http://www.sublimelayers.com/2016/10/some-musings-on-retracts.html) ) and thinking about it I would love to see both attempts suggested/explained by @DerEchteLanger  and @Ghostkeeper be implemented in some way or the other as especially on a bowden system like the Ultimaker I can see room for improvements.

Regarding Adaptive Retraction distances: I think its a good idea to have different settings/behaviours for different situations. As I don't know how useful it will be in the end I would probably just start with two settings, therefore I choose the wording "Split Retraction Settings". 
The split point I wouldn't define by the length of the next travel move as IMHO this isn't the influencing factor for oozing. I would rather choose the last value of material extruded/s of the last move as this should be reflecting the nozzle pressure (-> blobs, oozing). 

I do like the idea of retractions/part of the retractions happening while the nozzle travels even though I don't think that the assumption of a steady underpressure is completely true. 
IMHO it's  useful because the nozzle doesn't stop at the retraction point and therefore reduces also blobs. Maybe it can be mixed. For expample a short retraction 1mm and slow retraction until 6mm. I would limit the steady retraction to a value. 

@Ghostkeeper you said that ultimaker is already experimenting. Does this mean we can find some form of new settings for retraction in Cura's experimental section in a soonish release? Or maybe a plugin?

Ghostkeeper, 2018-07-06:
> @Ghostkeeper you said that ultimaker is already experimenting. Does this mean we can find some form of new settings for retraction in Cura's experimental section in a soonish release? Or maybe a plugin?

No, the experiment was stopped. It was part of an effort to reduce oozing in extremely long travel moves for a then-unreleased printer which is now called the Ultimaker S5 (Ultimaker's latest printer). I think it was not deemed important enough on our most popular print cores, because the small diameter cores such as 0.4mm have this way less than the larger ones.

Ghostkeeper, 2018-09-28:
Our project manager put this off of our planning. We won't get time to work on this any time soon.

Thisismydigitalself, 2019-05-03:
I am now printing a large model using PETG. i notice the same issue and I agree - This is a major issue.
I hope this is being handled or there is a solution/workaround for this. for small models i retarct 3mm for PETG@245C and it is more than enough but now the nozzle travels like 220mm across the model and i see filament oozing and ruining the next wall printed. I can't say I understand Cura's algorithm which is somewhat mysterious why it needs to jump so many times across the model from side to side - this contribues greatly to the poor outcome.

smartavionics, 2019-05-03:
>  I can't say I understand Cura's algorithm which is somewhat mysterious why it needs to jump so many times across the model from side to side - this contribues greatly to the poor outcome.

If you were kind enough to provide the project (.3mf) file perhaps we could understand what you are referring to?

Ghostkeeper, 2019-05-06:
> I can't say I understand Cura's algorithm which is somewhat mysterious why it needs to jump so many times across the model from side to side - this contribues greatly to the poor outcome.

Cura's algorithm for path ordering is in the basis just plain old nearest neighbour: After printing the previous line, look for the next line that still needs to be printed. It does this type by type, part by part under some constraints (which can be read here: https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#printing-order).

Mushmuch, 2019-12-13:
We need adaptive retraction distance. Try to print the rhino from Fab365 at 140% : no oozing in the first 3cms then when the areas are separated by more than 5cm there are zits and blobs and string everywhere :( Please.

[image]

loneFunction, 2020-02-04:
Along with adaptive retraction that is dependent on travel distance, there needs to be an adjustment to priming along with it. I have many issues with underprimed lines after travel moves. It doesn't seem to be from oozing either, since I don't get many strings. It may have something to do with my printer (Qidi X-Max) or the makerbot MK7 derived hotend design.

I think an ideal solution would be an initial quick retract (current retract speed) for the first x travel distance, followed by a slower retract amount based on travel, then prime scaled by travel distance.

There was a plugin that allowed for a scalable retraction prime which would address the last part of this. It does not work with Cura 4+ for some reason.
https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime

If someone could get that working or point me in the direction of why it's no longer compatible it would go a long way in helping my print quality.