Ultimaker/Cura #3454: Error: Print job larger in version 3.2.1 for ultimaker3 extended

recarens, 2018-03-08:
With curaEngine 3.2.1 I created a gcode correctly for Ultimaker3 extended with the default parameters of de ultimaker3_extended.def.json and the printer shows me: "This print job is larger than the printer's build volume." and if i load the same gcode from a Cura2.7 it loads correctly and i can print it.

recarens, 2018-03-08:
For example, is imposible is bigger than the base the object.

[image]

recarens, 2018-03-08:
Loading it from the usb :
[image]

closed 2018-03-09 by recarens

Ghostkeeper, 2018-04-03:
This was a bug in command line slicing that was fixed in 3.3.