Ultimaker/Cura #3455: Spiralize zseam = retract at layer change

JJJpnt, 2018-03-08:
Cura 3.2.1

Windows

Latest NVidia driver on a GTX970

Dagoma Discoeasy200 (set tu custom fdm printer)

I've seen quite a few Issue Post that seems to talk about this exact problem but didn't really know where to post so I'll made a new one : I also got a Z Seam doing a spiralized object. Actually I found very easily what was causing it : Retract at layer change was activated, and it triggered the retract at each "layer change" even if in vase mode. Deactivating it actually solved the issue. I guess a simple and definitive solution to automatically deactivate retract at layer change when using spiralized contours.

smartavionics, 2018-03-08:
You're absolutely right, a fix that does just that has recently been accepted. See https://github.com/Ultimaker/CuraEngine/pull/694.

JJJpnt, 2018-03-08:
Oh sorry I didn't see that ! I'll have to step up my searching before posting next time I open a ticket... 
Thanks for that :)

closed 2018-03-08 by JJJpnt

smartavionics, 2018-03-08:
No problem, thanks for taking the time to raise the issue (and also for closing it promptly!)