Ultimaker/Cura #3457: [3.2.1] Camera stream not visible after upgrade

cdrose, 2018-03-08:
**Application Version**
3.2.1

**Platform**
Windows

**Display Driver**
Intel HD Graphics 4000 20.19.15.4835

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
After upgrade to 3.2.1 the camera stream is no longer visible in the Monitor tab. The stream is available on http://<ip_address>:8080/?action=stream

ianpaschal, 2018-03-09:
Could you please share your log file so we can look for possible causes?

cdrose, 2018-03-12:
[stderr.log](https://github.com/Ultimaker/Cura/files/1804464/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/1804465/stdout.log)

Legor, 2018-04-05:
Same problem here

peteros85, 2018-04-05:
I have the same problem, Running Ultimaker 3 Extended over our company network, occasionally you will get a still from the camera but most of the time it wont load at all.

cdrose, 2018-04-05:
Good to see its not just me!

I just did a clean install of Cura 3.2.1 on a machine that has never had Cura installed before and the issue is the same.

peteros85, 2018-04-17:
Is this ever going to get looked in to? Its a big issue for me

cdrose, 2018-04-26:
Just upgraded to 3.3.1 and this issue appears to be fixed. Although I also did a firmware update on the printer at the same time so I am not sure which of the two actions solved the problem.

closed 2018-05-04 by ianpaschal

fieldOfView, 2018-05-04:
@peteros85, is it also fixed for you?

Legor, 2018-05-15:
Problem for us is solved with the latest version 3.3.1

ed8metzler, 2022-10-08:
Same problem on Octoprint 1.8.4 and Cura 4.13.1

fieldOfView, 2022-11-16:
@ed8metzler, for problems with the OctoPrint plugin, please see https://github.com/fieldofview/cura-octoprintplugin/issues