Ultimaker/Cura #3458: Enhanced combing : move over printed walls when there is no combing path -

MB3Y, 2018-03-08:
**Application Version**
3.2

**Platform**
Win10

**Steps to Reproduce**
Please see #3092 

**Actual Results**
When walls come close to eachother the insetted area in which the nozzle is forced to move for the combing feature, can be pinched off and result in separated areas. Today the nozzle is moving from one area to the next in a straight line, with retraction. The straight line can go from any area to any other area. The result can be a retracted move through "free air".

**Expected results**
When enabling combing we want to avoid 1. retraction and 2. stringing and/or moves over "free air". These goals are currently not met when walls are too close one to another.

**Additional Information**
**Suggestion**
Can combing be enhanced with a "bridge/link/narrowpath" feature, where "bridges/links/narrowpaths" are defined in between loose areas. Such a connection between such areas is to be determined as a printpath already followed during the printing of this layer. Keep filament retracted during a travel over such narrowpath. One could even "echo" such paths from previous layers several layers up to the current layer in an attempt to predict the best connection - or the least disturbing stringing/oozing path?

Trail at Hashuan
[image]

smartavionics, 2018-03-08:
A problem with your suggestion is that the inevitable dribble that occurs when combing will end up on the wall which could produce blobbing or other artifacts. Indeed, there is another issue raised (sorry, I don't have it's ref) which complains about combing down long thin walls producing bad results.

Personally, I think the existing combing implementation has so many issues that it should be rewritten from scratch. I'm trying to work up the energy and enthusiasm to tackle it but it's really daunting!

MB3Y, 2018-03-10:
> A problem with your suggestion is that the inevitable dribble that occurs when combing will end up on the wall
I don't agree with that statement although I understand the reasoning. When moving over a narrowpath linking wall, you should still do a retract. If one still has a dribble... yeah... then (s)he should probably try coasting on top ;)

Ghostkeeper, 2018-04-04:
We're considering adjusting the collision walls for combing because moving halfway alongside/through a wall is probably still better than moving through it orthogonally. That's basically this issue.

MB3Y, 2020-12-08:
In my opinion this one is closed - ref to https://github.com/Ultimaker/Cura/issues/3092

closed 2020-12-08 by MB3Y