Ultimaker/Cura #3459: Z stays at layer1 hight for next 2 layers @3.2.1

elilesw, 2018-03-09:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2018-03-09:
Unless you provide some evidence (the gcode and stl files would be lovely) we simply won't believe you!

elilesw, 2018-03-09:
Yes I dont blame You for that. I guess the issue is due to an other wrong setting but I can not find it.

elilesw, 2018-03-09:
[plantzip.zip](https://github.com/Ultimaker/Cura/files/1796710/plantzip.zip)

smartavionics, 2018-03-09:
Thanks for the files. When I look at the gcode, the layers are printed at the expected heights (0.2, 0.4, 0.6 and so on). Here's a view showing what's happening part way through the second layer, z is 0.4 as you would expect:

[image]

elilesw, 2018-03-09:
Yes I agree it looks ok when looking in layer view, but when printing it is wrong.

The nozzle is cutting down into the 1st layer when printing layer 2 & 3 and the display is still showing  Z0.2
Infill is in 2 diffrent directions for these layers so it is not a repeat. 
Did You try to print?

smartavionics, 2018-03-09:
Just noticed, in your start code you have this sequence:

>G1 Z2.0 F230 ;Move the head up 2mm
G92 E0       ;Prime the extruder
G1 F200 E1.51
G92 Z2.45 E0

Which says go to z = 2.0 and then tell the firmware that z is 2.45, that cannot end well!

elilesw, 2018-03-10:
Yes I owe You an explanation for "G92 Z2.45"
I zero Z against a PB switch mounted through my alu buildplate at X0 Y0 and it switches at Z0.45.

Thank You for showing me the direction. You came very close. The offset was already compensated for in Marlin. 

Could there be a check for an actual Z value change for a new layer?

smartavionics, 2018-03-10:
> Could there be a check for an actual Z value change for a new layer?

Cura doesn't notice that you have set z using g92 in the start code so as far as it is concerned the buildplate is located at z=0 and all the z values it uses are positive.

BTW - there is a Cura plugin that lets you set a z offset.

closed 2018-03-10 by elilesw