Ultimaker/Cura #3461: All materials show up and some of them crash Cura

maukcc, 2018-03-09:
With the latest 3.2.99 all materials show up in a "has_machine_materials" definition machine and some of them even crash Cura.

Could someone please make a read.me file with mandatory values on relations between the definition file and others files like:
- variant files
- quality files
- material files
It feels like it has been made deliberately extremely complicated to get this to work properly, and with every version of Cura, links between them seem to change.

In this read.me should be:

in definition file what do the following do and how to use them properly?
version
has_machine_quality
has_materials
has_machine_materials
has_variant_materials
has_variants
preferred_variant
preferred_material
preferred_quality

In quality files what do the following do and how should the values be formulated:
material = 
variant = 

In material files
is there a specific <version></version> and what does this do?
does  <diameter></diameter> do anything in the latest version?

ChrisTerBeke, 2018-03-12:
@LipuFei @ianpaschal is there some documentation available for this already?

ianpaschal, 2018-03-12:
@ChrisTerBeke I don't think so. I mean this is more a question for the material's department...

Ghostkeeper, 2018-04-06:
It's not really a question for Ultimaker's material department. This is about the structure of our resources and how to link them properly so that certain resources are available for certain printers. It's not deliberately made complex but it's made complex by the difficult requirements that Cura has had to meet. Sadly these requirements didn't include to keep it easy to learn how to create new machine definitions, so yes it's pretty complex.

closed 2019-01-07 by nallath