Ultimaker/Cura #3462: problème de supports

lionelcemoi, 2018-03-09:
bonjour,
CURA ne fait pas tous les supports nécessaires pour l'impression.

**Application Version**
CURA 14-7

**Platform**
windows10-64

**Printer**
ANET A8

**Steps to Reproduce**
aucun

**Actual Results**
impression impossible

**Expected results**
impression impossible

**Additional Information**
http://prntscr.com/iow92j
[image]

ChrisTerBeke, 2018-03-09:
https://translate.google.com/translate?hl=en&sl=en&tl=fr&u=https%3A%2F%2Fgithub.com%2Fdaid%2FLegacyCura

ianpaschal, 2018-03-09:
Bonjour, nous ne travaillons plus sur la version 14.x (legacy) de Cura. Veuillez passer à au moins 2.7, mais de préférence la dernière version (3.2.1). C'est dans le [website de Ultimaker](https://ultimaker.com/en/products/ultimaker-cura-software).

Et s'il vous plaît utiliser l'anglais sur github. 😉

ianpaschal, 2018-03-09:
J'ai fermer pour maintenu mais si il ne marche pas avec le nouveau version de Cura, répondez encore.

closed 2018-03-09 by ianpaschal

lionelcemoi, 2018-03-09:
sorry, but I don't like CURA 3.2

lionelcemoi, 2018-03-09:
CURA 14-7 is simple, easy to use, even for someone like me who does not speak English. Its interface is clear and well done.
CURA 3.2.1 is too complicated, too bushy. There are too many things on the screen, you end up not understanding anything. By cons it is in French, but it does not facilitate its use so far.
This is my personal opinion

ianpaschal, 2018-03-09:
You can hide most parts of Cura 3.2.1 if you want. In fact you can very easily hide all settings except the few you use. Let me know if you'd like help with adjusting that.

But if not, then you will need to fix the problem yourself. The repository for legacy Cura is here: https://github.com/daid/LegacyCura

lionelcemoi, 2018-03-09:
Thanks, that's nice.
First of all, my printer is an unmodified ANET A8.
Should I use RepRap or Marlin?
I tell you that I am a beginner in 3d printing. I am a 3D programmer but not a printer

ianpaschal, 2018-03-09:
I'm not an expert on ANET A8 but the default firmware is really terrible according to a colleague who has an ANET A6 at home. He says with Marlin on his ANET A6 works well with Cura. But that's a recent version.

Unfortunately, if you are not a programmer, you must rely on the tools others build. In this case, people like you find issues, tell us, we fix them, and release a new version. This has happened many times and the result over the years is 3.2.1.

You cannot expect problems to be fixed but also not upgrade to the versions in which they are fixed. So if you are not a programmer, I recommend you upgrade. Cura 3.2.1 is very powerful, very stable, and definitely one of the best pieces of slicing software ever created. If you are worried that the interface is too cluttered, it's VERY easy to turn off all settings except the ones you want to see. If you upgrade I will tell you how.

lionelcemoi, 2018-03-09:
I am aware that it takes time to get a preforming software and that despite all efforts, there will always be dissatisfied.
For ANET A8, I have no problem, it is a machine that works very well with its original firmware.
My only problem is the configuration of CURA.
Speaking of CURA, an improvement that would be nice would be to add or remove media, much like in easyprint (which I use professionally)

ianpaschal, 2018-03-09:
> I am aware that it takes time to get a preforming software and that despite all efforts, there will always be dissatisfied.

That's not really my point. It's just that your bug is very likely fixed a long time ago, but you need to upgrade to get the fix.

> My only problem is the configuration of CURA.

For this part, normally you have to provide steps of what you did and what did not work but you have said "aucun". In this case though, even if you provided the steps, we would not try to solve the bug because no one in the company uses these very old versions of the software and we cannot make a separate branch of development just for one person.

lionelcemoi, 2018-03-09:
I put "none" because I did not find any solution to my support problem

ianpaschal, 2018-03-09:
Ah but if you had the solution why open an issue? "Steps to Reproduce" means how can we cause the problem to occur. So tell what you were doing, what settings you used, give a link to the model, etc. This way we can try to follow the same steps and watch what goes wrong and then make a fix. If we cannot recreate the problem on our computer, we cannot fix it.

But in any case, we will not be working on a Cura 14 bug anyway. If you need help making Cura 3.2.1 more streamlined, let me know. Otherwise I will not follow this thread anymore.

Bon weekend!

lionelcemoi, 2018-03-10:
CURA 3.2.1 being reserved for specialists, I went to Repetier.
I keep CURA 14.7 for easy prints

lionelcemoi, 2018-03-10:
[image]