Ultimaker/Cura #3465: [FR] Branch / Bridging Supports

dot-bob, 2018-03-10:
I would like to discuss an idea what I might coin as branch, bridging, or overhang supports.  It is where a support is attaches itself to the side of a print face in one or two small spots and grows out to support features above it.  

Thus it is similar to how a branch grows from a tree or can be used like the bark to protect the tissue underneath.  The supports can climb without worrying about poor overhang surface quality supporting the surface we care about.  The supports then can be broken off and the small imperfection on the side of the print can be easily repaired.  This will help with tall prints that supports to the bed would become tall and unstable or use a large amount of material.

ianpaschal, 2018-03-10:
Have you tried the tree support experimental feature?

dot-bob, 2018-03-10:
Yes I have tried tree supports and is a great feature.  The difference is that tree supports require a flat face to start from and the overhangs of the supports themselves are conservative.  

The idea of this support is that it can attach to vertical faces or possibly mild overhangs of the print.  Thus the supports begin printing just before the feature that needs support.  You would also be able to specify the angle of the supports and the distance before the feature to begin supports as this may be printer and material dependent.  When manual supports come along you would position them on the side of a face with the mouse. 

It rides on the concept that you don't care as much about the underside surface quality of the supports so you can use extreme overhangs with supports that may droop as long as the support has built up and ceased drooping before it comes in contact with the feature it needs to support.

Ghostkeeper, 2018-04-04:
The way you are describing it, I think you mean something like this (with the cyan bits being support):

[image]

The support would be growing from the nearby vertical faces of the print rather than going all the way to the bottom, thus requiring less material and time at the cost of a worse surface finish at the side of the print where it needs to attach.

Am I understanding this correctly?

dot-bob, 2018-04-06:
Correct...  You could use conventional support or even branch supports.  Perhaps specify the allowable overhang angle of the supports.  Then a user could use an aggressive angle and it is ok if the supports have droop as long as the supported area is unharmed.  People could use an overhang test to see what their printer is capable of.

Perhaps also have an option to specify the density or number of the connection points to the side of the model.  You could also define the location as random, on a edge / corner, or a specific location.

Ghostkeeper, 2018-04-30:
Tree support does this. Normal support doesn't do this yet, but for normal support it's quite a lot more difficult due to the nature of the shapes that they work with.

You could implement this with some polygon magic like this:
* Offset the model areas and support areas such that the X/Y distance is bridged.
* Join the model areas with the support areas.
* Inset the model+support areas with the same distance so that the original support+model perimeter is back (bug: this closes small holes in support too, so it effectively adds to the Support Join Distance).
* For all model+support area parts that overlap any of the model areas: Inset the model+support areas with a certain inset such that the desired support angle is achieved: tan(support_angle) * layer_height.
* Subtract the model areas with X/Y offset from the model+support areas.

It'd work, but it would take some computation. There's 8 linear polygon operations in there.

closed 2023-07-13 by github-actions[bot]