Ultimaker/Cura #3468: [3.2.1]  Loading shorter model first followed by taller model cuts the taller model to shorter model height

vernieg, 2018-03-10:
[3.2.1]  Loading shorter model first followed by taller model cuts the taller model to shorter model height

**Application Version**
3.2.1

**Platform**
Linuxmint 8.3

**Printer**
Anycubic i3 Mega

**Steps to Reproduce**
Load a shorter height model file on the build plate. Load another taller model file. Go to Layer view.

**Actual Results**
Maximum printable layers is only that of the shorter model.  Loading a taller model first then shorter second will makes all printable.
This has been replicated many times with .stl or .obj files.

**Expected results**
All models should be printable at their full height.

**Additional Information**
[image]

smartavionics, 2018-03-10:
Hi, could you please make those models available so I can try and recreate the problem? Thanks.

ianpaschal, 2018-03-10:
Are you using adaptive layers? If so, this is a known bug we're working on.

vernieg, 2018-03-10:
Yes, just noticed it now, this only trigger on using adaptive layers. Thank you.

ianpaschal, 2018-03-10:
Duplicate of #3318.

ianpaschal, 2018-03-10:
This will be fixed in 3.3.

closed 2018-03-10 by ianpaschal