Ultimaker/Cura #3469: [3.2.1] Opening single file loads recently loaded files

Aeysol, 2018-03-10:
**Application Version**
3.2.1

**Platform**
Arch Linux
```
[Aeysol@chemistry ~]$ uname -a
Linux chemistry 4.15.7-1-ARCH #1 SMP PREEMPT Wed Feb 28 19:01:57 UTC 2018 x86_64 GNU/Linux
```
**Qt**
5.10.1

**PyQt**
5.10

**Steps to Reproduce**
1. Open STL file in Cura
2. Open another STL file in Cura

**Actual Results**
Opening a single file opens all previously loaded files.

**Expected results**
Only the selected file is added to the workspace without adding all previously loaded files.

**Additional Information**
I've patched the file Cura.qml to include a  `console.log(fileUrlList);` after [line 719](https://github.com/Ultimaker/Cura/blob/7312ed8e3c32b8923744487b7bb4f39f0925d252/resources/qml/Cura.qml#L719) to show the file list that is returned by the `FileDialog`.

After opening the first file:
```
qml: file:///home/Aeysol/Downloads/test.stl
```

Second file:
```
qml: file:///home/Aeysol/Downloads/test.stl,file:///home/Aeysol/Downloads/test2.stl
```

Third file:
```
qml: file:///home/Aeysol/Downloads/test.stl,file:///home/Aeysol/Downloads/test2.stl,file:///home/Aeysol/Downloads/test3.stl
```

So after opening the third file I have a total amount of 6 models in my workspace.

ChrisTerBeke, 2018-03-13:
I'm not able to reproduce this (on latest MacOS). Maybe it's an (Arch) Linux specific issue?

Ghostkeeper, 2018-04-12:
I've actually had this issue for a while when opening files in Cura via the Open File dialogue, using Ubuntu 17.04. Instead I was dragging the model into the build plate as alternative. Eventually Canonical fixed it for me. I don't remember exactly when.

This leads me to agree that it's probably specific to Linux, or at least upstream somewhere where we can't fix it. I also can't reproduce this issue any more now.

closed 2018-04-12 by Ghostkeeper