Ultimaker/Cura #3475: [3.2.1] Bigger layer spent more filament (!)

mtmello2003, 2018-03-11:
Win10 x64

I doesn´t make sense for me:  When I set layer height to 0.2 (for example), I gave xx meters and xx grammar of filament needed to print but if I increase the layer to 0.3, the ammount will increase too...

Bigger layer will generate less total layer needed to print the item then, will consume less filament.

[image]

[image]

fieldOfView, 2018-03-12:
> Bigger layer will generate less total layer needed to print the item then, will consume less filament.

No, because each of the thicker layers will extrude more material than the thinner layers. With all other parameters (such as line width) set to the same value, the total amount of material for the print should be in the same ballpark. The brim is going to take more material to print with a bigger layer height though.

ianpaschal, 2018-03-12:
Yup, thanks @fieldOfView. @mtmello2003: It may seem strange but is expected.

ChrisTerBeke, 2018-03-12:
Doesn't seem like a bug, closing.

closed 2018-03-12 by ChrisTerBeke