Ultimaker/Cura #3476: Sync SimulationView with Octoprint currently printing layer

t0mahawk-jb, 2018-03-12:
**Application Version**
3.2.1

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**
Nvidia

**Printer**
N/A

**Steps to Reproduce**
N/A

**Actual Results**
N/A

**Expected results**
N/A

**Additional Information**
Feature request for SimulationView: Sync SimulationView to Octoprint plugin status to show simulation of currently printing layer.  This provides a similar feature Repetier's live view.  Although not perfectly synced to the currently running GCode, would at least keep the simulation in sync on an at layer level.

fieldOfView, 2018-03-12:
While it could be "fun" to have this, it is all but trivial. The SimulationView uses layer data it gets from CuraEngine, not the actual gcode data. 

The OctoPrint API includes which line is currently being sent to the connected printer ("filepos", http://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job), but there is no accurate way to relate that back to the layer data that similation view uses.

In addition to that, printing with OctoPrint does not block your use of Cura, so you can explicitly continue working with Cura to prepare other prints in Cura. This deletes the layer data, and replaces it with new layer data that could very well have no relation to the currently ongoing print.

diegopradogesto, 2018-03-12:
Thanks @fieldOfView for the really good explanation. Probably in this case it's better to implement the feature in Octoprint but not in Cura.
BTW, I implemented the latest version of the SimulationView and I was planning in doing a more realistic simulation, using the speed and the length of each segment to simulate according to them (at this moment each line is drawn at the "same speed" and not taking into account the length). Of course it won't be perfect for all the printers since other parameters (such as acceleration) may also affect.

fieldOfView, 2018-03-12:
Well, no, I would implement it in Cura instead of OctoPrint (so USB and otherwise connected printers can also benefit), but I was pointing out why it is not a simple undertaking. It would require PrinterOutputDevices to keep a copy of the layerdata (as well as the gcode data). I will not have time for that soon though.

ChrisTerBeke, 2018-03-12:
Given this is (likely) not going to be added to Cura, maybe close the ticket and mark as 'wont' fix'? That'll keep the backlog clean.

closed 2018-03-12 by ianpaschal