Ultimaker/Cura #3480: Request: Printing "one at a time"

mtmello2003, 2018-03-12:
Can be great if there is an option to execute a custom gcode when cura end to print a piece and go to the next one when the option "one at a time" is set.

[image]

I wish to add M600 for example, to change filament collor but the post script gcode do it only by layer, not by pieces at a time...

BagelOrb, 2018-03-16:
Because one at a time printing is fundamentally a lot more difficult for dual extrusion machines, this feature won't have priority for ultimaker.

One at a time mode is disabled for dual extrusion printers.

Ghostkeeper, 2018-04-11:
You can still print with only one of the nozzles on a dual extrusion printer though :stuck_out_tongue: 

In the post-processing script Filament Change you can give layer numbers that go beyond the first model though. If you have two models of 200 layers each, you should be able to indicate that it needs to pause at layer 75 and at layer 275, just like how Layer View indicates it.

coreyfaure, 2018-07-11:
I would also like to see this feature, as I would like to add an M300 beep code after every individually printed object.

closed 2023-07-13 by github-actions[bot]