Ultimaker/Cura #3481: Support Direction (Angle)

darkdvd, 2018-03-13:
I posted something on the forum Ultimaker:
https://community.ultimaker.com/topic/21946-support-direction-angle/

ChrisTerBeke, 2018-03-13:
It seems indeed that there is no infill direction parameter for support infill. I've flagged this ticket with the correct tags.

darkdvd, 2018-03-13:
Merci Chris !

darkdvd, 2018-03-14:
I suggest adding a function to the supports management in CURA: "Support Direction".

By default, the supports are printed along the X and Y axes, regardless of the angle of the object.
(Note that the object is positioned by default along the X and Y axes as well).

In certain cases, the geometry of the object to be printed poses a problem during printing: this is particularly the case for objects intended to be molded (The walls of these objects generally have angles of 2 ° in order to optimize demolding, sometimes it is the design of the piece that uses very tight angles ...).

I took as an example the piece that hides the linear bearings of the UM3: this piece is intended to be made of molded plastic (1905-H).

[image]

The front of this object has an angle of 2 °.
In my example, I print the object vertically to limit the amount of supports. 

[image]

Problem # 1: this wall at 2 ° will be badly printed because the displacement of the head will not be precise enough to restore the wall which must be perfectly rectilinear.

Solution # 1: rotate the object at 45 °. The angle of this wall will be no longer 2 ° but 47 °, which changes everything. 
  
[image]

Problem # 2: the supports remain positioned on the X and Y axes and they overflow the object and the printing will necessarily be a failure.

Solution # 2: give the possibility to change the angle of the supports.
In this case, if the object is turned at 45 °, it is necessary to rotate the supports by 45 °, which is not possible at present (or I missed something ....). 
Note #1 : The object contained in the STL has been modified to be printed vertically to limit the number of supports.
Note #2 : I use zigzag supports because they are (far) the best support in terms of finishing and easier to remove.
Note #3 : I am french...

ChrisTerBeke, 2018-03-14:
I think by default we could even align the support infill and normal infill directions, with the option to override that.

darkdvd, 2018-03-14:
Good idea.

darkdvd, 2018-03-14:
The "liberty" to choose the direction is a good option too.

By default, the value "align Infill to Object" is not checked.
If checked, the value "0" can be modified to any other value...

darkdvd, 2018-03-27:
So... ?

ChrisTerBeke, 2018-03-27:
Cura's backlog is super cramped, it takes a while until things are picked up. In the mean time, any developer is free to make a pull request for this.

@Appesteijn is this something you think is useful?

darkdvd, 2018-03-27:
Thank you @ChrisTerBeke !

Note : not only "useful" but "necessary"

ChrisTerBeke, 2018-03-27:
Haha, that's always a matter of opinion 😄

darkdvd, 2018-03-27:
(I'm not sure my last sentence was in English...)

Appesteijn, 2018-03-29:
I think it is very useful, but it will also depend on how much work this is if (and when) we can pick this up. I hope some devs can mingle in this discussion :)

Ghostkeeper, 2018-04-25:
The work on this is probably very little. We just have to fill this variable properly instead of hard-coding it to 0:
https://github.com/Ultimaker/CuraEngine/blob/2e8d4b0eca8702a9833349f4e7b306610f80e852/src/FffGcodeWriter.cpp#L2064

Ghostkeeper, 2018-07-04:
Even still, our project manager just removed this feature from our planning. We won't be able to spend time on it soon.

fieldOfView, 2018-07-04:
I'll see what I can do.

fieldOfView, 2018-08-08:
See https://github.com/Ultimaker/CuraEngine/pull/819 and https://github.com/Ultimaker/Cura/pull/4212

closed 2018-08-31 by alekseisasin