Ultimaker/Cura #3482: [3.2.1] Brim only on outside not working

idotuchman, 2018-03-13:
**Application Version**

Cura 3.2.1

**Platform**

Window 10

**Steps to Reproduce**

Using this [model](https://www.dropbox.com/s/hya51tbyf7g8tvh/Motor%20Holder.stl?dl=0), when I select brim only on the outside, Cura makes brims around the inside pillars in the model.

**Actual Results**

[image]
**Expected results**

[image]

**Additional Information**

ChrisTerBeke, 2018-03-13:
I have the same thing when printing lego-brick type models, so that checks out. Not sure though if it's supposed to work like this (technically it's still the 'outside' of a 3D solid) or it's indeed a bug. Maybe @Ghostkeeper or @nallath can shed some light on that?

ianpaschal, 2018-03-15:
Hehehe... Cura team recognizes a Lego brick when they see one.

I don't think this is the expected behavior though because the alternative definition of an "inside brim" would just be the floor.

So I'm going to label this as a bug but we can remove it if we get a better explanation.

BagelOrb, 2018-03-16:
It seems to me that the brim is on the *outside* of the pillars which are on the inside. Seems to work as intended.

The setting expressly puts brim on the outside of all shapes. If those shapes are contained within other shapes then those other shapes have to get a brim in the inside as well, due to implementation limitations. (Otherwise the outside brim on the inner objects could overlap with the outer shape)

ChrisTerBeke, 2018-03-16:
That's what I thought as well, let's close this then. If we want to change the behavior, we can either create a new ticket with a description of the new behavior or re-open this one.

closed 2018-03-16 by ChrisTerBeke

ianpaschal, 2018-03-16:
@BagelOrb Isn't the issue though that if the pillars inside get a brim then in fact all edges get a brim? To me "outside only" would imply that only outer-most polygon of the first layer should get a brim, which should also avoid brims running into each other within the object.

Ghostkeeper, 2018-04-13:
> Isn't the issue though that if the pillars inside get a brim then in fact all edges get a brim? To me "outside only" would imply that only outer-most polygon of the first layer should get a brim, which should also avoid brims running into each other within the object.

The way it was intended is to still have a brim on the outside of parts that happen to be surrounded by something else. The function of the brim is to provide better bed adhesion. Whether an object happens to be surrounded by another object doesn't matter for its bed adhesion. This particular setting is meant to save time because the brim in a hole has less of an effect than the brim on the outside of a perimeter.

However due to an implementation limit we couldn't make it work for holes that have some other part inside it. I think that can be fixed but it'll take some effort and it won't save all that much time/material in total.

smartavionics, 2018-04-24:
Another example, I would not expect to see brim on the inside of the outer ring:

[image]