Ultimaker/Cura #3488: Strange printing artifact introduced after 2.7

biscuitlad, 2018-03-13:
**Application Version**

2.7 and 3.2
**Platform**

Ubuntu Artful
**Qt**

Not needed
**PyQt**

Not needed
**Display Driver**

Not needed
**Printer**

Vertex K8400 Dual Nozzle
**Steps to Reproduce**

Print object using version 2.7, then using 3.2
**Actual Results**

Strange lip introduced on printed object from 3.2
**Expected results**

Objects should look identical (same STL)
**Additional Information**

[image]

ChrisTerBeke, 2018-03-13:
Was there a difference in settings for the printer settings of this printer between these two Cura versions?

biscuitlad, 2018-03-13:
Relative extrusion, but I don't think it's relevant.

This issue occurs on one side only, it's like a little step. I tried
rotating the model and reprinting it, but I can't get rid of it.

It's not there in the layer view, so it seems to be caused by the gcode
instructions to the printer.

ChrisTerBeke, 2018-03-13:
In that case it doesn't sound like a Cura issue to me. Could you share the G-codes so we can double check?

biscuitlad, 2018-03-13:
Sadly I can't, as I have managed to delete the 2.7 version and I've
upgraded to 3.2 with a new install of artful. I just noticed as I had a
previous print of the same object.

I've attached the gcode for the 3.2 version, and the STL, if that helps.

Ghostkeeper, 2018-04-12:
Maybe it has something to do with filling small gaps? If it goes faster through the part then you'll get more warping.

We did change a lot of things since 2.7 was released last summer. Also defaults for settings, so some settings will still be different from what you had in 2.7. We might need a better tuned profile for your printer, but without access to that printer itself there is not much we can do.

Unless there is something you can point out in the g-code that is wrong? I tried slicing your bracket in Cura 3.3 but I don't see anything that is obviously wrong with it.

ianpaschal, 2018-05-29:
Haven't heard back in 6+ weeks, so I don't think we're going to be able to address this issue.

closed 2018-05-29 by ianpaschal