Ultimaker/Cura #3489: [3.1+] Mistranslation of "Hollow out objects" in French

MajorSlug, 2018-03-13:
**Application Version**
Since 3.1 at least 

**Platform**
Windows

**Actual Results**
"Éviter les objets" means "avoid objects", not "hollow out objects"

**Expected results**
"Évider les objets"

ChrisTerBeke, 2018-03-13:
@Ghostkeeper Something you might wanna pass on to the translators?

7Guillaume7, 2018-03-23:
Hello,

Cura
Version : 3.21
Windows

Actual Results :
"Touche" et "touché par" Ils définissent une action du corps humain.
They define an action of the human body.

Expected results :

Touche->Dirige
Touché par -> Est dirigé par

Ghostkeeper, 2018-04-17:
Thanks guys. I've fixed these for the release of 3.3 [here](https://github.com/Ultimaker/Cura/commit/0f39b36bd6bf9aee10e6bbc830614a857b3a5100) and [here](https://github.com/Ultimaker/Cura/commit/1b173e8119ed384ada9942baf952beb97dc91569).

Keep 'em coming if you see more mistakes.

commit 0f39b36b (2018-04-17): Correction to French translation of 'Hollow Out Objects'

Fixes #3489.
commit 1b173e81 (2018-04-17): Correct translation of 'Affects'

Apparently 'touché' means something more like a human touching it, rather than something affecting something else.

Fixes #3489.
closed 2018-04-17 by Ghostkeeper